Ultimaker/Cura #8336: Some grouped models unslicable

BagelOrb, 2020-09-07:
**Application version**
4.6, 4.7, master

**Platform**
Windows 10, Linux Mint 19 Cinnamon

**Printer**
S5

**Reproduction steps**
1. Load attached model
2. Duplicate
3. Position models so that both are printable
3. Group them

**Screenshot(s)**
[image]

**Actual results**
Models are both yellowtaped aka marked as unslicable

**Expected results**
Both models have the material color and are slicable

**Project file**
[UMS5_vase_bottom.3mf.renamed.zip](https://github.com/Ultimaker/Cura/files/5178327/UMS5_vase_bottom.3mf.renamed.zip)

**Log file**
```
2020-09-07 15:05:50,900 - WARNING - [MainThread] cura.UI.PrintInformation.setBaseName [366]: Unsupported Mime Type Database file extension 
2020-09-07 15:05:50,906 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: <UM.Operations.AddSceneNodeOperation.AddSceneNodeOperation object at 0x7fe5cc49f4e0>, took 3ms
2020-09-07 15:05:50,909 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [188]: Socket state changed to Listening
2020-09-07 15:05:50,934 - INFO - [MainThread] UM.Backend.Backend.startEngine [90]: Started engine process: /home/t.kuipers/Development/CuraApp/bin/CuraEngine
2020-09-07 15:05:50,935 - DEBUG - [MainThread] UM.Backend.Backend._backendLog [110]: [Backend] Calling engine with: ['/home/t.kuipers/Development/CuraApp/bin/CuraEngine', 'connect', '127.0.0.1:49674', '']
2020-09-07 15:05:51,013 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2020-09-07 15:05:51,013 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] Cura_SteamEngine version master
2020-09-07 15:05:51,014 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] Copyright (C) 2020 Ultimaker
2020-09-07 15:05:51,014 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2020-09-07 15:05:51,014 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] This program is free software: you can redistribute it and/or modify
2020-09-07 15:05:51,015 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] it under the terms of the GNU Affero General Public License as published by
2020-09-07 15:05:51,015 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] the Free Software Foundation, either version 3 of the License, or
2020-09-07 15:05:51,015 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] (at your option) any later version.
2020-09-07 15:05:51,015 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2020-09-07 15:05:51,016 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] This program is distributed in the hope that it will be useful,
2020-09-07 15:05:51,016 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] but WITHOUT ANY WARRANTY; without even the implied warranty of
2020-09-07 15:05:51,016 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
2020-09-07 15:05:51,017 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] GNU Affero General Public License for more details.
2020-09-07 15:05:51,017 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2020-09-07 15:05:51,018 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] You should have received a copy of the GNU Affero General Public License
2020-09-07 15:05:51,018 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] along with this program.  If not, see <http://www.gnu.org/licenses/>.
2020-09-07 15:05:51,286 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._checkStack [159]: Need to check for errors again. Discard the current progress and reschedule a check.
2020-09-07 15:05:51,460 - DEBUG - [MainThread] UM.Mesh.MeshData.approximateConvexHull [506]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.04311227798461914 seconds. 2111 input vertices. 2111 output vertices.
2020-09-07 15:05:51,518 - DEBUG - [MainThread] UM.Mesh.MeshData.approximateConvexHull [506]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.05703306198120117 seconds. 2111 input vertices. 2111 output vertices.
2020-09-07 15:05:51,519 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [33]: Updating CustomQualityProfilesDropDownMenuModel.
2020-09-07 15:05:51,520 - DEBUG - [MainThread] cura.Machines.Models.QualityProfilesDropDownMenuModel._update [62]: Updating QualityProfilesDropDownMenuModel.
2020-09-07 15:05:51,538 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [192]: Socket state changed to Connected
2020-09-07 15:05:51,539 - DEBUG - [MainThread] UM.Backend.Backend._onSocketStateChanged [181]: Backend connected on port 49674
2020-09-07 15:05:51,960 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [155]: New error check scheduled.
2020-09-07 15:05:52,439 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [214]: Error check finished, result = False, time = 0.5s
2020-09-07 15:05:54,461 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: TranslateOp.(node=<CuraSceneNode object: 'Group #1'>,trans.=<0.0,0.488,0.0>), took 15ms
2020-09-07 15:05:54,487 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: TranslateOp.(node=<CuraSceneNode object: 'Group #1'>,trans.=<0.0,4.89,0.0>), took 9ms
2020-09-07 15:05:54,514 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: TranslateOp.(node=<CuraSceneNode object: 'Group #1'>,trans.=<0.0,4.4,0.0>), took 12ms
2020-09-07 15:05:54,531 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: TranslateOp.(node=<CuraSceneNode object: 'Group #1'>,trans.=<0.0,4.89,0.0>), took 5ms
2020-09-07 15:05:54,551 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: TranslateOp.(node=<CuraSceneNode object: 'Group #1'>,trans.=<0.0,2.45,0.0>), took 9ms
2020-09-07 15:05:54,580 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: TranslateOp.(node=<CuraSceneNode object: 'Group #1'>,trans.=<0.0,3.42,0.0>), took 9ms
2020-09-07 15:05:54,596 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: TranslateOp.(node=<CuraSceneNode object: 'Group #1'>,trans.=<0.0,2.45,0.0>), took 5ms
2020-09-07 15:05:54,616 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: TranslateOp.(node=<CuraSceneNode object: 'Group #1'>,trans.=<0.0,1.47,0.0>), took 11ms
2020-09-07 15:05:54,644 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: TranslateOp.(node=<CuraSceneNode object: 'Group #1'>,trans.=<0.0,0.985,0.0>), took 7ms
2020-09-07 15:06:01,225 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
```

konskarm, 2020-09-07:
I can reproduce. It seems to happen only when the first extruder is disabled. I'll bring it up for discussion.

konskarm, 2020-09-09:
Devs see ticket CURA-7710.

commit a613a707 (2021-11-05): Don't disable group nodes if it's extruder is disabled

Group nodes don't really have an extruder, so exclude them from that check

CURA-7710
Fixes #8336
closed 2021-11-05 by nallath