Ultimaker/Cura #8337: Z slider in Preview in percent?

globules-io, 2020-09-07:
**Is your feature request related to a problem? Please describe.**
When looking at monitor, print is expressed in %, but slider is in Z units. So I can't go back to check where we at on the preview.

**Describe the solution you'd like**
Be able to display % or Z units. Or even in the Monitor view, have a button at the bottom (like pause) [preview]  and on click preview, show Preview view at this stage. Real time update would be nice too.

Ghostkeeper, 2020-09-10:
See also https://github.com/Ultimaker/Cura/issues/7576 and https://github.com/Ultimaker/Cura/issues/7683 for different ideas of what the layer slider should show.

We could make it switch to % of the total amount of layers. Only this % would not be the same as the % you see while printing via USB or via the network. Those are based on time, by measuring the time that has elapsed since the start of the print vs. the total printing time. Currently the front-end has no information about time estimates per layer; it would need to receive this from CuraEngine, so that's something that needs to be implemented first then. Otherwise the % will be widely off, since most prints designed for 3D printing have wider bases than tops (to prevent the need for support) and are printed with a thick bottom skin which is printed slowly.

Ghostkeeper, 2020-09-10:
We've discussed this with the team and decided to not give this priority over our other issues now, unless there is a more pressing need for it.
I think my ideal situation would be that the contents of that layer slider are more configurable in the preferences screen. However we also need to consider that the layer view slider may not be the best place to display metadata information about layers either.

globules-io, 2020-09-10:
Hey thanks for looking in to this. I understand the issues that comes with this feature.
I had one remark about the elapsed time. I feel it very bizarre that it is not just a simple timer, that starts with pressing start, pause/resume with pause/resume and ends when the print is fully done. I don't understand why this would have to be calculated by the way? Time that passes... it just a timer?

Ghostkeeper, 2020-09-11:
Layer view needs to work before actually starting the print, so there is nothing to measure the time of.

globules-io, 2020-09-11:
What? What about when the end user presses PRINT ?

Ghostkeeper, 2020-09-15:
You wouldn't expect the layer view Z slider to change when you press print, right?

Or are you requesting features entirely unrelated to the Z slider in Preview?

globules-io, 2020-09-15:
Yes you are misunderstanding, won't waste more on this.

closed 2020-09-15 by globules-io