Ultimaker/Cura #8338: Crashes on Loading STL

edm00se, 2020-09-07:
**Application version**
4.7.1

**Platform**
macOS Catalina 10.15.6

**Printer**
Creality CR-10

**Reproduction steps**
1. Upgraded from 4.7.0 to 4.7.1 (which I thought would fix the bug)
2. Dragged STL file into Cura (previously working in 4.6.2, not in 4.7.0)
3. Crash

**Screenshot(s)**
[image]

**Actual results**
Cura crashes 😢 

**Expected results**
Cura should display my newly imported STL file (this is one I've sliced with Cura previously).

**Project file**
I can't get that far sadly.

**Log file**
Here's [my `cura.log` saved in a gist](https://gist.github.com/edm00se/bfd2bd619f8b5c975fc7c0725381a6d9).

**Additional information**
Based on #8328 and #8230, I thought for sure I was in the clear; sadly this doesn't seem to be the case.

nallath, 2020-09-07:
Could you copy your configuration folder? It could be that 4.7.0 messed up something so that it still doesn't work.

edm00se, 2020-09-07:
I've got it zipped up, assuming you mean the dir from `~/Library/Application Support/cura`. I omitted the older 4.1, 4.3, and 4.5 sub directories.

https://www.dropbox.com/s/ahrqpzbufmszyko/cura.zip?dl=0

nallath, 2020-09-07:
Yeah, it's as I feared. 4.7.0 stored your start g-code incorrectly. 

You need to change the Creality+CR-10+%232_settings.inst.cfg file to:
```

[general]
version = 4
name = Creality CR-10 #2_settings
definition = creality_cr10

[metadata]
type = definition_changes
setting_version = 15

[values]
extruders_enabled_count = 0
layer_height = 0
machine_end_gcode = G91 ;Relative positioning
	G1 E-30.0 F2700 ;Retract a bit
	G1 E-2 Z0.2 F2400 ;Retract and raise Z
	G1 X5 Y5 F3000 ;Wipe out
	G1 Z10 ;Raise Z more
	G90 ;Absolute positionning
	
	G1 X0 Y{machine_depth} ;Present print
	M106 S0 ;Turn-off fan
	M104 S0 ;Turn-off hotend
	M140 S0 ;Turn-off bed
	
	M84 X Y E ;Disable all steppers but Z
machine_extruder_count = 0
machine_start_gcode = ; Script based on an original created by tjjfvi (https://github.com/tjjfvi)
	; An up-to-date version of the tjjfvi's original script can be found
	; here:  https://csi.t6.fyi/
	; Note - This script will only work in Cura V4.2 and above!
	; --- Global Settings
	;
	; magic_mesh_surface_mode = {magic_mesh_surface_mode}
	;
	; retraction_amount = 0
	; retraction_enable = {retraction_enable}
	;
	; retraction_hop = 0
	; retraction_hop_enabled = {retraction_hop_enabled}
	;
	; retraction_prime_speed = 0
	; retraction_retract_speed = 0
	; retraction_speed = 0
	; smooth_spiralized_contours = {smooth_spiralized_contours}
	;
	; speed_travel = 0
	; speed_z_hop = 0

```

edm00se, 2020-09-07:
Oof yeah. I’m guessing that happened when I first tried with 4.7.0? Either way I can give this a try later tonight and report back.

nallath, 2020-09-07:
> Oof yeah. I’m guessing that happened when I first tried with 4.7.0? Either way I can give this a go later tonight and report back.

Yeah. Unfortunately, this isn't something we can detect / recover automatically. The shitty thing is, we can see that >500 people have had this issue, so I hope we don't have to manually fix all of those...

edm00se, 2020-09-08:
To update, I tried copying and pasting the above contents for `Creality+CR-10+%232_settings.inst.cfg` and it didn't do the trick for me. I saved [a screen shot and a copy of my `cura.log` (gist)](https://gist.github.com/edm00se/802529f4d925912b09e3c9874e44e36f) from that crash if it would be helpful.

I did wind up trying the alternate approach you mentioned, which was to delete the 4.7 directory (I renamed it to something else that shouldn't be used) and was happy to find that it worked as expected when importing an STL file.

So, to compare both the correctly working copy of things and the mangled copy, as well as the proposed above copy, I've got that here as screen shots from FileMerge.

Mangled vs the good copy of things (created without 4.7 folder with 4.7.1, does the migration from my prior 4.6.2 settings and works):
[image]

The proposed above vs the same good copy of things:
[image]

It looks like not just the comments from the octolapse dump of settings mostly comes across, but still missed my actual start gcode in addition to a `machine_extruder_count=0` being added.

I'm able to use Cura again at 4.7.1, so thank you. Also if there's something else you can gain from my setup, settings, or logs, please let me know. Also it seems I can give any remedial actions you'd like to try a go, as I have a saved copy of a mangled 4.7 environment.

nallath, 2020-09-09:
Ah, right, it also mangled the order. I didn't notice that when manually patching the start-gcode. 

Thanks for looking into it (and for the detailed write up). If more people would write bug reports like this, our life would be so much easier!

closed 2020-09-09 by nallath

Ghostkeeper, 2020-09-10:
I've also seen [reports where 4.7.1 did fix it for people that were previously experiencing the bug](https://www.reddit.com/r/Cura/comments/ij1r7b/cura_47_cant_import_any_models/). I think that it automatically gets fixed as long as the user doesn't modify their machine settings in 4.7.0. Because if these settings are not modified, the file doesn't get re-saved and never actually gets corrupt (only interpreted wrong in 4.7.0).