Ultimaker/Cura #8340: [4.7.0] Corrupt Configuration

petarvucetin, 2020-09-07:
**Application version**
(The version of the application this issue occurs with.)
4.7.0
**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Windows 10
**Printer**
(Which printer was selected in Cura?)
Ender 5

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)
Start a program. Go through wizard. Load print. Clone Dynamic settings. Change settings. Close program. Start program. 
"Your configuration seems to be corrupt. Something seems to be wrong with following profiles:
- Creality Ender 5
- creality_ender5_hatchbox
Would you like to resdet to factory defaults?"

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]

**Actual results**
(What happens after the above steps have been followed.)
Requires a reset of all configuration settings. Creates duplicate Ender 5 profile with #2 at the end

**Expected results**
(What should happen after the above steps have been followed.)
Should start without resetting.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)
Problem is in startup 

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues
 to find the log file to upload, or copy a relevant snippet from it.)
[cura.log](https://github.com/Ultimaker/Cura/files/5184295/cura.log)

**Additional information**
(Extra information relevant to the issue.)
All the files in the cura folder
[cura.zip](https://github.com/Ultimaker/Cura/files/5184308/cura.zip)

nallath, 2020-09-07:
Upgrade to 4.7.1, this was fixed there.

closed 2020-09-07 by nallath

petarvucetin, 2020-09-07:
Installed new version. Kept my config files. Same problem.

[image]

petarvucetin, 2020-09-07:
Have to delete and re-create all the settings and printers to make it go away.

nallath, 2020-09-07:
You can just delete the creality_ender_5 files from your configuration folder.

fieldOfView, 2020-09-07:
I can recommend installing the Start Optimiser plugin from the Marketplace; it adds a button to the Configuration errors dialog to disable only the corrupted configuration files instead of resetting everything.