Ultimaker/Cura #8342: Ironing despite 0 Top Layers

ameuret, 2020-09-08:
**Application version**
4.7.0

**Platform**
Linux

**Printer**
Custom (Monoprice Select Mini)

**Reproduction steps**
1. Add model
2. Add mesh modifier model (for overlaps and cutting mesh)
3. Set mesh modifier setting for Top Layers (and Bottom Layers) to 0
4. Enable Ironing (in general settings, not modifier mesh settings)
5. Slice and observe that an ironing pass is generated on unexisting top layer

I checked that when disabling the general _Enable Ironing_ setting, the spurious pass disappears.

**Screenshot(s)**

[image]

**Actual results**
Ironing pass is generated.

**Expected results**
No ironing pass when no layer to iron exists.

**Project file**
https://groupeactime-my.sharepoint.com/:u:/g/personal/arnaud_actime_biz/EcxK6WZSgddBvNyVxsksaCcBfdVV_pq_8sqBdMQlYonsTw?e=8Icew0

rburema, 2020-09-09:
Also doesn't work the other way around (ironing on for modifier mesh, ironing off in general).

Malmik, 2021-01-23:
Just experienced this:
(You can see that there is NO top layer, while there is ironing, becouse you coan look throught the ironing)
[image]
[image]

MalicT, 2021-11-20:
Just had this happen with mine too, was doing a 2 wall box for calibration of flow, with 0 top layers, because of ironing option being checked had it try to do a top layer, took a bit to figure what was causing this

Ghostkeeper, 2021-11-29:
Shouldn't you then, like, disable ironing instead?

Ironing doesn't specify that it only works on skin layers. In fact, it normally goes over all of the walls too. That's part of what makes it smooth out the surface well.

MalicT, 2021-11-29:
that is the point of the bug report, it should disable trying to iron a layer that does not exist, it should only iron the tops of the walls, not try to do the entire flat area, since top layer was set to 0

tastyratz, 2023-01-10:
I noticed this recently and it seems like it's still happening. If Ironing is left on it will iron the top phantom layer, not calculated/actual layer.

Even when you set iron only highest layer it will try and iron the top of the object as it thinks that's the topmost layer. This makes ironing objects printed without a top for use as a vase impossible.

MariMakes, 2023-02-09:
We've added a ticket to the backlog to improve this. 
For internal reference CURA-10271

Thanks for the report! 👍

daniels220, 2023-05-16:
I would add that disabling ironing for an area of skin that _does_ exist, using an infill mesh, also doesn't work, and there's no reason for it not to. Using a cutting mesh allows it to work, but generates spurious outer walls when trying to affect only part of a layer.

GregValiant, 2024-11-20:
Is this still an issue in current versions of Cura (5.8.0 and up)? Can this be closed?

daniels220, 2024-11-21:
The behavior seems to be unchanged, that is, Cura will iron wherever the model is present, even if no top layer is present. This is true regardless of whether modifier meshes are involved, and conversely, using a _cutting_ mesh to disable ironing altogether works properly—you can disable ironing across only part of the model this way. However, this means that the boundary between ironing/no ironing will always have a wall, and this wall cannot be made any thinner than the outside walls of the part as a whole (you could try by introducing another modifier mesh but this then introduces its own wall). So this is pretty limiting.

GregValiant, 2024-11-22:
Thanks for getting back.  I'll leave this open.

GregValiant, 2024-11-23:
Thanks for getting back.  I'll leave it open.

daniels220, 2024-12-07:
@GregValiant hmm...do you have to remove the "needs info" tag for the bot to not auto-close it?

MalicT, 2024-12-07:
yeah this is still an existing issue, and should not have been closed

GregValiant, 2024-12-07:
The bot should have removed the label when you responded and it should not have closed.  That's a Github thing.
I used the label on a lot of these older reports to thin the herd.  So far this is the only one that was closed erroneously.