Ultimaker/Cura #8343: Abnormal perimeter slicing in Cura 4.7 and 4.7.1

Whiskaya, 2020-09-08:
**Application version**
4.7 and 4.7.1

**Platform**
macOS Catalina

**Printer**
Creality Ender 5

**Reproduction steps**
1. Slice a Benchy in Cura 4.7 and print it.

**Screenshot(s)**
Slicing
[image]

Reviewing GCODE (notice the blobs on the bow)
[image]

**Actual results**
The perimeters get these strange dots coming out.

**Expected results**
Get smooth perimeters.

**Project file**
[GCODE.zip](https://github.com/Ultimaker/Cura/files/5185752/GCODE.zip)

**Log file**
(Irrelevant)

**Additional information**
1. It's not the STL: MatterControl, PrusaSlicer and Cura 4.6.2 work fine (even using the same profile).
2. Doesn't happen with every STL (I just noticed in this one because the Benchy was particularly awful)
3. Tried the stock profile for Ender 5, same result.
[image]
[image]

jellespijker, 2020-09-08:
Probably duplicate copy of #8321

Whiskaya, 2020-09-08:
> Probably duplicate of #8321 

Indeed.
Sorry, seems like I didn't look for it long enough.

closed 2020-09-08 by Whiskaya