Ultimaker/Cura #8346: 4.7.1 Doesn't show weight of material after slicing e.g. ( 0gms )

casdsys, 2020-09-08:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

konskarm, 2020-09-08:
@casdsys could you please fill in the given template? It will help us find and solve the issue.

rburema, 2020-09-09:
What @konskarm said.

But also, is the weight for the material set properly?

closed 2020-09-22 by no-response[bot]