Ultimaker/Cura #8347: Flyingbear ghost 5 not listed in printer library but stated in updated features.

casdsys, 2020-09-08:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

konskarm, 2020-09-08:
@casdsys Please fill in the template when reporting a bug.

Also, the Flying Bear Ghost 5 has not been merged yet (https://github.com/Ultimaker/Cura/pull/8200). Once it is merged, it will show up in the printer list in the next Cura version. Finally, what do you mean it is mentioned in updated features?

rburema, 2020-09-09:
@konskarm I thought they could mean the Changelog we include, but there's no mention of a flying bear there as far as I can see?

closed 2020-09-22 by no-response[bot]