Ultimaker/Cura #8348: MSI installer for Cura 4.7.1: Product is already installed

Sebazzz, 2020-09-08:
**Application version**
4.7.0 to 4.7.1

**Platform**
Windows 10

**Printer**
N/A

**Reproduction steps**
1. Install Cura 4.7.0 using the MSI installer
2. Attempt to upgrade Cura 4.7.1 using MSI installer

**Screenshot(s)**
[image]

**Actual results**
A message is shown:

> Another version of this product is already installed. Installation of this version cannot continue. To configure or remove the existing version of this product, use Add/Remove Programs on the Control Panel.

**Expected results**
Cura 4.7.1 should be installed, without uninstalling Cura 4.7.0.

**Project file**
N/A

**Log file**
[Windows Installer log file](https://github.com/Ultimaker/Cura/files/5190288/cura_470.log)

**Additional information**
It appears you forgot to update the Windows Installer "ProductCode" for the new version of Cura. Re-uploading the installer with a new productcode should resolve it.

rburema, 2020-09-09:
For updating to a newer patch version via MSI: This is a known issue at the moment. We'll be working on this for future releases. We've made internal ticket CURA-7662 for this task.

For the more general problem of 'multiple Cura's side-by-side on Windows'; we advise to install via the exe installer.

Ghostkeeper, 2020-09-10:
I thought this was just a limitation of the MSI installer that we have to live with? The only way of fixing that would be to give a unique name to the installer for every release (e.g. `Cura4.7`), but that would prevent the uninstallation of old versions via command line.

rburema, 2020-09-10:
@Ghostkeeper  Yes, when it comes to major and minor releases. Patch level updates should be able to _overwrite_ the older installation. Which is a bit different from 'install this version next to the other one and potentially keep both'.

Ghostkeeper, 2020-09-11:
The expected results in this issue are to keep both, though:

> Cura 4.7.1 should be installed, without uninstalling Cura 4.7.0.

rburema, 2020-09-11:
> The expected results in this issue are to keep both, though:

Yes, which is why I said _"For the more general problem of 'multiple Cura's side-by-side on Windows'; we advise to install via the exe installer."_ :-)

That might've not been a good way of putting it?

Sebazzz, 2020-09-13:
> 
> 
> The expected results in this issue are to keep both, though:
> 
> > Cura 4.7.1 should be installed, without uninstalling Cura 4.7.0.

Oh sorry, I mean to say "Cura 4.7.1 should be installed, without *needing to manually* uninstalling Cura 4.7.0."

closed 2024-12-08 by github-actions[bot]