Ultimaker/Cura #8349: [4.7.1] Support Blocker Not Blocking

Snargola, 2020-09-08:
**Application version**
4.7.1

**Platform**
Windows 10 Home, Intel Core i5-7200U

**Printer**
CR-10s

**Reproduction steps**
1) Loaded model.
2) Orientated model.
3) Placed support blocker.
4) Clicked, 'slice'.
5) Looked at preview.
6) Frowned.
7) Repeated steps 1-6 multiple times, varying step 2, many, many times.
8) Came here to post this.

**Screenshot(s)**
[image]
[image]
[image]
[image]
[image]
[image]

**Actual results**
An unexpected result. Supports are not blocked, as requested.

**Expected results**
I had hopped that supports in the area selected would be blocked. They are not.

**Project file**
https://www.dropbox.com/s/f9m45ed7chs6bso/ttyt_copper_robbie_sitting_body.3mf?dl=0

**Log file**
No crash but, I can still add the log file later, if you would like.

**Additional information**
Nothing further to add.

EeleDahc, 2020-09-09:
Not working in 4.7.0 on my machine either. Same results.   AMD Ryzen 5 3600.  16gb ram.  Just upgraded from 4.6. Uninstalled 4.6 but kept data.

nallath, 2020-09-09:
The project file that you provided doesn't have any support blockers in it. When I add one myself, it blocks the support as intended. 

Could you re-check the project file and see if it's the right one?

Snargola, 2020-09-09:
> 
> 
> The project file that you provided doesn't have any support blockers in it. When I add one myself, it blocks the support as intended.
> 
> Could you re-check the project file and see if it's the right one?

I have downgraded to Cura 4.6.2 since I posted this however, the problem still persists. I'll post a DropBox link to the project file, gcode and profile I am using.
[image]
Project file
-----------
https://www.dropbox.com/s/lqkj3ezqch7d2h2/robbie_sitting_body.3mf?dl=0

gcode
------
https://www.dropbox.com/s/jp79gois5a9zbed/ttyt_copper_robbie_sitting_body.gcode?dl=0

profile
------
https://www.dropbox.com/s/w1v8srx0f29vz8x/ttyt_copper_robbie_sitting_body.curaprofile?dl=0

Thank you for your help so far.

nallath, 2020-09-10:
You have the `support join distance` set to 2mm and the support blocker itself is also 2 mm. This, combined with the `support horizontal expansion` of 0.5mm, means that the supports are initially removed but then merged back together again.

closed 2020-09-10 by nallath