Ultimaker/Cura #8350: How to get rid of M109 after tool change in cura

sTsx, 2020-09-08:
HI
I'm using Cura 4.7.1 with custom printer setup.  I have a Makertech Proforge 2S Dual Extrusion Setup. It's not in the list of printers to select from, so I entered the Specs, Start and End Gcodes manually. 

Issue is Cura adds a M109 after every tool change.  I set All print temperatures to 230. As well as the standby temperature to 230, if I don't set the standby it sets to 120 automatically.  The M109 causes the print head to rest on the printed part for awhile before moving on.  Is there a way to eliminate the M109 command after the tool changes?
[image]

Ghostkeeper, 2020-09-10:
You can add a Search and Replace post-processing script that replaces `M109 S\d+\n` with nothing (enabling regex). This removes all `M109` commands from the g-code though so you'd have to put something in the start g-code that doesn't match that, which you currently already have in your start g-code.

In 4.6 there was an option that said that nozzles share a heater that would prevent these commands if enabled. That's what you're really requesting here. This is hidden in 4.7 though because it wasn't a working solution for most printers. See https://github.com/Ultimaker/Cura/issues/8148 for a discussion on that issue. I'll close this in order to keep the discussion together.

closed 2020-09-10 by Ghostkeeper

sTsx, 2020-09-10:
I read through #8148 It deals with retraction on a shared heater. I’m using dual extruder with Dual heater. I’ll try the search and replace that you suggested. Do you have a suggestion on an alternative for the M109 in the start gcode though?  I haven’t come across one yet.

Ghostkeeper, 2020-09-15:
You could comment it out with a semicolon in front of the line?