Ultimaker/Cura #8352: [4.7] Incorrect g-code generated?

jtrmal, 2020-09-09:
**Application version 4.7**
(The version of the application this issue occurs with.)

**Platform MacosX**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer Ender 5 Pro**
(Which printer was selected in Cura?)

**Reproduction steps**
The g-code generated does not produce expected results -- I'm not sure if it's problem with the slicer or the printer, but with I got a fine print with a slightly different number of walls

**Screenshot(s)**
[image]

**Actual results**
Seems like part of the segment was not printed correctly

**Expected results**
... would be printed correctly

**Project file**
See the project and the g-code  in the following zip file
[Archive.zip](https://github.com/Ultimaker/Cura/files/5194333/Archive.zip)

jtrmal, 2020-09-09:
Never mind, probably not a problem in Cura -- could not reproduce for this g-code (could reproduce for a g-code where it didn't happen yesterday). IE issue somewhere else...

closed 2020-09-09 by jtrmal

Ghostkeeper, 2020-09-10:
If we get a problem like this, we also need to know what's wrong about the g-code. We can't reproduce what's wrong with the print because that is dependent on the printer and environment.

jtrmal, 2020-09-10:
Yeah, I thought the  g-code contains some operation "move diagonally"
instead of "up" and "left", which I'd assumed would be obvious.
Anyway, not an issue with cura, sorry for bothering...
y.