Ultimaker/Cura #8353: 3 Dual extrusion issues: Not purging 2nd color on first layer, Not printing last color on final layer, Not unloading tool when done.

B-rad888, 2020-09-09:
So I'm having a few issues/bug's  with dual extrusion.  I start from the beginning to the end of the process. 

1.  When I print with the prime tower, the priming does not start until layer 3. I'm using a 2 in 1 out nozzle. this does not leave enough time for the nozzle to purge. Hence a discolored bottom on the second color. This is really not good for anything where the bottom is the top of a finished project. See pictures in **screenshots**

2. Next when I get to the final layer the second color doesn't print. The nozzle just hovers over the area like it should, but does not deposit any filament. The extruder has no movement during the final pass of the final color. Therefore it does not get printed. See pictures in **screenshots**

3. Finally, tool 0 and 1 (extruder 1 and 2) have start and end gcode. I would like after the last color and layer is finished for it to do its unload sequence (extruder 2 end g-code). That way nothing is in the filament path. However after the last color pretends to print (the black color) it stays loaded in the nozzle. This means if I went to print a new file tool 0 (white) would ram into tool 1 (black). So I have to manually remove filament from tool 1 (extruder 2) each time I want to start a new print.  

**Application version**
Cura 4.7.0 and 4.7.1

**Platform**
Windows 10

**Printer**
Custom printer based on FDM printer

**Reproduction steps**
1. Import print
2. Print

**Screenshot(s)**
[image]
[image]

**Actual results**
First layer doesn't purge, Last color on last layer doesn't print
 

**Expected results**
Purge every layer, and print every layer

**Project file**

G-code file I'm using to test (generated by Cura)
[Test Gcode.zip](https://github.com/Ultimaker/Cura/files/5195017/Test.Gcode.zip)

STL I'm using to test
[Test stl.zip](https://github.com/Ultimaker/Cura/files/5195292/Test.stl.zip)

**Additional information**
Custom Printer made by referencing "inherits": "fdmprinter", with custom extruders, variants, and quality. As instructed by @Ghostkeeper . I don't think it has to do with the custom printer. However I can test other pre-built printer profiles because i'm using a modded Ender 3 pro with a 2 in 1 out nozzle.

B-rad888, 2020-09-09:
Issue 2 is solved: It turns out on the last color the temperature started dropping for some reason. I thought I had cura fooled from switching temperature by setting the standby temp of the material to 200c (same as the print temp). Worked good until the last color on the last layer. My solution: enable "machine_extruders_share_heater". It works now. 

Issue 3 is solved: I read through the G-code. For some reason on the last color the extruder start g-code is present but not the extruder end g-code. I'm not sure why that is set up like that. My solution: add extruder end g-code to machine end gcode. 

Issue 1: I think this is an oversight of cura. In issue #8148  this was mentioned along with the share heater setting. I don't see what is wrong with the with the "machine_extruders_share_heater" setting. According to issue #5770 and @smartavionics  the setting is only meant to do the following:

> I have put a checkbox into the machine settings dialog that specifies that the heater is shared between the extruders. When this box is checked, the previous extruder temperature is not changed at all after the new extruder has been selected and its temperature set. If that is how it should behave, I can post a build later today for you to test.

I think the issue is what is pictured above. Cura does not utilize the prime tower for priming the second color until layer 3 on the tower. This is an issue for not just dual extrusion with a raft, but also insufficient priming as in my case with a short brim or skirt. It doesn't make sense to me why the priming tower would not utilize the first layer for priming the second color just like it does in layer 3 and every switch thereafter. I believe if it did, this would solve both issues with dual extrusion.

I think a fix to the first layer of the prime tower would be the right solution. I also think that enabling the shared heater option as it was, would also be beneficial. I don't see how removing a feature and functionality will make things better. Instead I would suggest making the small (but probably tedious) change to the prime tower. Either way please don't remove "machine_extruders_share_heater" from fdmprinter.def.json. It's working well for me.

Ghostkeeper, 2020-09-11:
> This is an issue for not just dual extrusion with a raft, but also insufficient priming as in my case with a short brim or skirt.

For that there is the [Skirt/Brim Minimum Length](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/platform_adhesion/skirt_brim_minimal_length.md) setting. It will add additional contours to the skirt to ensure that it's primed enough.

The rest of the remaining issues are being discussed in #8148. It's better to keep the discussion together in one issue. If `machine_extruders_share_heater` is working well for you, please add to that issue how it's possible that for you the filament is not colliding with previously loaded filament.

closed 2020-09-11 by Ghostkeeper

B-rad888, 2020-09-11:
Thanks @Ghostkeeper . I decided it is better to just use a purge bucket. The prime tower is not efficient for dual extrusion. Plus Less waste and more efficient to use a purge bucket. I'm finishing my design now. I'll post all my findings once the system is fully operational. I'll let everyone know how to set up the filament so it does not collide in the other post in the meantime.

kryvoshyia, 2021-05-30:
Did you find a solution with first layer?

Ghostkeeper, 2021-06-07:
@kryvoshyia A similar bug arose in Cura 4.9.0, which is fixed in 4.9.1. So if you're seeing that, please update!

nickcalthrop, 2021-09-04:
Any solution yet on second extruder not purgeing in first layer??? 

i am having this problem in Cura 4,10

Ghostkeeper, 2021-09-14:
Yes Nickcalthrop. The setting Extruders Share Nozzle was added to resolve the issue this is a duplicate of. Please continue discussion in the original thread.