Ultimaker/Cura #8355: In custom print settings, arrow handles on section headings don't open/close properly

jacix, 2020-09-09:
**Application version**
4.7.1

**Platform**
MacOS 19.6.0 Darwin Kernel Version 19.6.0

**Printer**
Creality Ender 3 Pro and Creality CR10S Pro

**Reproduction steps**
1. In the "Prepare" workspace, go to custom print settings and open any of the section headings.  The arrows on the right don't reorient.  This started with version 4.7. 

**Screenshot(s)**
After starting Cura without any project loaded:
[image]

Loading a recent project then clearing everything by selecting "new project":
[image]

**Actual results**
The arrows don't open or close.

**Expected results**
The arrows should open and close.

**Project file**
This happens whether I have a project file loaded or not.

**Log file**
Nothing appears in the log file when I see this behavior.  

**Additional information**
(Extra information relevant to the issue.)

konskarm, 2020-09-10:
I can reproduce this issue in windows. Thanks for the report.

Ghostkeeper, 2020-09-10:
Duplicate of #8283.

closed 2020-09-10 by Ghostkeeper