Ultimaker/Cura #8356: Profiles still wont show up

choppy420, 2020-09-09:
I Redownloaded Cura 4.71 and when I tried to add profiles, it is not showing up.  Not sure if it's just my Artillery that is having the problem but my Creality Cr-10s had the same issues

konskarm, 2020-09-10:
Hi @choppy420. Can you give a little bit more information? (see the [template](https://github.com/Ultimaker/Cura/issues/new?assignees=&labels=Type%3A+Bug&template=bug-report.md&title=))

TheMaxx2804, 2020-09-12:
Hi,
i have the same problem as mentioned here and in similar issues.
I am new to Cura and installed it the first time.
It seems that this problem is depending on the filament diameter size in the profiles and which is configured with the printer / extruder settings.

I found the template for reporting issues and hope i filled it correctly...

**Application version**
4.7.1

**Platform**
Windows 10 64bit
Version 2004 Build 19041.508

**Printer**
Prusa i3 MK3s

**Reproduction steps**
1. New (first) installation of Cura
2. Created an Ultimaker Account for data synchronizing
3. Tried to import material profiles from market place
3. Choose one where i have filament (BASF InnoPET)
   Cura told me that the import was sucessfully
4. restart of Cura
    ...but i could not see the new material / profile

**Screenshot(s)**
n.a.

**Actual results**
Not showing the imported material in list

**Expected results**
Showing up the new material in list

**Project file**
n.a.

**Log file**
n.a.

**Additional information**
I downloaded the material profile manually and reckognized that the profiles were for 2.85mm filament.
I changed the filament size in the xml profile manually to 1.75 and reimported the material profile which then was showing up in the list.

TheModernWizard, 2020-09-14:
Can confirm. I have the same issue. (same reproduction instructions and other template information as The Maxx2804 but with an Ender 3.

nallath, 2020-09-15:
@TheModernWizard and @TheMaxx2804 That is because the materials on the marketplace are only designed for 2.85mm and Ultimaker printers.

TheModernWizard, 2020-09-15:
Oh, actually I didn't notice the 2.85 mm part it's just that my custom profiles stopped existing in Cura 4.7.

nallath, 2020-09-15:
> Oh, actually I didn't notice the 2.85 mm part it's just that my custom profiles stopped existing in Cura 4.7.

Did you check if https://github.com/Ultimaker/Cura/issues/8229 might be the issue of that?

TheModernWizard, 2020-09-15:
Wow that just might fix it. Thank you for the suggestion! I think everyone with this issue has the same problem outlined in #8229.

choppy420, 2020-09-16:
Thank you will try that out

closed 2020-10-01 by no-response[bot]