Ultimaker/Cura #8358: Overextrude a tiny bit at the start of each line while on the first layer

Fusseldieb, 2020-09-09:
**Is your feature request related to a problem? Please describe.**
Yes, my bowden machine doesn't like it so much when I print directly on glass. Even with the bed tweaked just perfectly, at the very start of the print sometimes filament doesn't come out just yet and it does build up ooze while printing in midair. Eventually the ooze gets sufficiently big and it adheres on the glass. A lot will say that I "just need to make it squeeze harder on the glass". While this is a workaround, it's not optimal. Another thing that I did is enabling Linear Advance and tweaked it just perfect. One thing that I routinely do is to loosen the extruder gear and jam a little bit of filament in and then let the gear go in its position again. This immediately causes perfect bed adhesion, because the extruder has now immediately pressure and filament to begin with.

**Describe the solution you'd like**
An option to overextrude a tiny bit at the start of each line while on the first layer (aka Initial/Bed layer) which should cause enough pressure buildup to start the line just fine.

**Describe alternatives you've considered**
Written above

**Affected users and/or printers**
Especially Bowden Printers printing on Glass, but could benefit other printer types as well.

**Additional context**
-

nallath, 2020-09-10:
I might be misreading your request wrong, but that is what I use the skirt / brim for.

Zogar89, 2020-09-10:
Cura doesn't have the option to apply an extrusion multiplier factor for the first layer? ==> Initial Layer Flow

Fusseldieb, 2020-09-10:
> I might be misreading your request wrong, but that is what I use the skirt / brim for.

The skirt is what I use, but it starts without any pressure in the nozzle and doesn't stick until seconds later, which sometimes gives failed prints.

> Cura doesn't have the option to apply an extrusion multiplier factor for the first layer? ==> Initial Layer Flow

I don't want to change the Flow of the whole layer, just the very start needs to have pressure.

nallath, 2020-09-10:
Have you tried Enable Prime Blob?

Fusseldieb, 2020-09-10:
> Have you tried Enable Prime Blob?

> This setting is for dual extrusion machines only.

Zogar89, 2020-09-10:
If you have a hole in the first line it is because by default the code before starting printing makes the bleed line to the left, when the line ends it says "retract a little". You would have to eliminate that line or minimize the retraction that is used to cut the bleed line.

You can see it in the printer settings, Gcode before you start printing.

Fusseldieb, 2020-09-10:
> If you have a hole in the first line it is because by default the code before starting printing makes the bleed line to the left, when the line ends it says "retract a little". You would have to [...]

I already defined a purging line at the start at the print, but this doesn't solve it completely. Also, if I don't retract at the end of the print, the nozzle oozes at the next start - also not a solution.

Ghostkeeper, 2020-09-11:
Essentially what you're asking for is the Retraction Extra Prime Amount setting, or perhaps more specifically a variant of that that's specific to the first layer.

But your problem just seems to be priming. There are a number of solutions for it, all of which have been mentioned here:
* Use a brim/skirt. Fusseldieb claims it doesn't seem to be a solution since the skirt lets go of the build plate, and then gets dragged along over the build plate ruining the print. I think the best solution to that would be to increase the number of skirt lines or the brim line width. It'll then get more chance to adhere to the build plate before the nozzle moves on to the actual print.
* Enable Prime Blob setting. This is only supported by the UM3 and later Ultimaker printers (which is why the support page for Ultimaker is technically correct). It's a bit misleading though since it also works with single-extrusion prints on those machines. It's got little to do with actual dual extrusion, just happens to be implemented in the newer printers of Ultimaker.
* Use the start g-code to prime. The prime line that Zognar mentions is specific to the start g-code of some printers (in particular the popular Ender 3, but also some others). I think this is also a very good solution.

If priming at the start of the print (either start g-code or skirt) doesn't solve the problem, I think Retraction Extra Prime Amount will, or else Initial Layer Height + bed levelling. Because I can think of 3 reasons why this could be a problem:
* Hysteresis and friction in the Bowden Tube, i.e. Retraction Extra Prime Amount, but that happens for every layer so there's no reason to have that specific for the first layer.
* Insufficient priming, i.e. prime using skirt or start g-code.
* The bed adhesion is insufficient, i.e. level the bed properly and use other bed adhesion techniques such as Initial Layer Flow, Initial Layer Height, etc.

Fusseldieb, 2020-09-11:
> * I think the best solution to that would be to increase the number of skirt lines or the brim line width. It'll then get more chance to adhere to the build plate before the nozzle moves on to the actual print.

I've done that. Problem is, it begins with so little pressure that it sometimes doesn't stick onto the build plate (due lack of pressure) and wrinkles up, forms a blob and that's it. I can cancel the print and start over. The problem is the small amount of pressure at the beginning. Once it wrinkles the ooze up, it has a difficult time sticking and even if it sticks, it leaves a blob behind and in the next skirt iteration it goes over that blob again, sometimes even carrying it a little bit.

> * Enable Prime Blob setting. This is only supported by the UM3 and later Ultimaker printers

I have a custom built Marlin 3D printer. That will likely not work.

> * Use the start g-code to prime.

As I have said earlier, when it primes at the beginning and stops, there will be pressure buildup in the tube and it *will* ooze. Those 2-3 seconds which it takes for the head to move from the prime location to the print position and the bed lifting is enough to cause ooze which makes skirting much harder, almost the same as I've mentioned above.

> I think Retraction Extra Prime Amount will

Maybe, but sadly it applies for every layer, which would overextrude every corner and details.

> or else Initial Layer Height + bed levelling

I already leveled the bed in a good height and it prints just fine, but the very start is a problem. This problem could be mitigated just by overextruding a tiny bit at the beginning of the print.

> Hysteresis and friction in the Bowden Tube, i.e. Retraction Extra Prime Amount, but that happens for every layer so there's no reason to have that specific for the first layer.

Stangely enough, it prints just beautifully after that initial squeeze that I do for it to stick to the bed. Only the first layer would need that.

> Insufficient priming, i.e. prime using skirt or start g-code.

Mentioned above, causes ooze by the time it actually starts to print. As immediate as it may sound, the head still needs to move into position and lower the bed. This takes 1-2 seconds.

> The bed adhesion is insufficient, i.e. level the bed properly

Already done several times. It sticks well and doesn't come lose while printing. It only needs a little initial "help", which could very well be automated.

Ghostkeeper, 2020-09-15:
Extrusion lines are always marked with `G1` by CuraEngine and travel moves with `G0`. So to find the first line with a regex you would simply need to search for `G1 `. The post-processing plug-in has a few helper function to help parse the `E` value from a g-code line.

It seems illogical to me that you'd get oozing after priming in the start g-code. You are retracting afterwards, right? The prime line is just as much of a line as any other line to the printer, so if this one oozes but other lines don't then something weird is going on with the prime line.

Fusseldieb, 2020-09-15:
> Extrusion lines are always marked with `G1` by CuraEngine and travel moves with `G0`. So to find the first line with a regex you would simply need to search for `G1 `.

Yes, I know, but the first line (or circle) is (obviously) splitted across a lot of G-Code lines and I would probably need overextrude for a certain length. I'm currently thinking on how to do it.

> The post-processing plug-in has a few helper function to help parse the `E` value from a g-code line.

Which one? I saw some python examples in the plugin folder, but for my "idea" which plugin is the best start there?

> It seems illogical to me

Yep, I also hate my printer sometimes. Bowden printers are often crap, and day after day I agree with that statement more. Especially tiny "dots" that Cura generates are utterly useless for Bowden printers, since it goes to a location, puts a "dot" of filament and leaves. Since the pressure isn't immediate, it just gets dragged around. I hate my printer so much (sometimes).

Ghostkeeper, 2020-09-17:
> Which one? I saw some python examples in the plugin folder, but for my "idea" which plugin is the best start there?

All of them. The post-processing scripts define a class that extends from the `Script` class. That class has this helper function available:

https://github.com/Ultimaker/Cura/blob/dd359da1cfeba7110791c98b2291af8d08cd98af/plugins/PostProcessingPlugin/Script.py#L117-L134

So from any post-processing script you can call `self.getValue("G1 X120 Y23.4 E42.6969", "E")` and this would return the `E` key of that line, 42.6969 in this case.

I think the [Search and Replace](https://github.com/Ultimaker/Cura/blob/master/plugins/PostProcessingPlugin/scripts/SearchAndReplace.py) script is the best example of how to write a post-processing script in general. It doesn't process any g-code though, so something closer to what you want to achieve can be found in the [Retract Continue](https://github.com/Ultimaker/Cura/blob/master/plugins/PostProcessingPlugin/scripts/RetractContinue.py) script, although that is a bit more complex.

Anon0ne, 2020-10-24:
I have had some luck with the fan coming on (mine is a 5015) to about 20% as soon as the purge/prime line is done to prevent the oozing. I have had oozing ever since upgrading to an all-metal hot-end and direct drive. It's just whatever is in the melt section flows out a tiny bit. The fan does help.

Otherwise do as the previous guy said and modify GCode with Post Processing scripts and make the line over-extrude. You should be able to do it via GCode with a little practice.