Ultimaker/Cura #8359: [4.7.1] Segmentation faults when trying to load larger STLs

TurboTurtle, 2020-09-09:
**Application version**
4.7.1

**Platform**
Fedora 32
kernel-5.8.4-200.fc32.x86_64
python3-3.8.5-5.fc32.x86_64
nvidia GTX 980

**Printer**
Snapmaker 2.0

**Reproduction steps**
1. Try to open a large-ish (~50+ MB) STL

**Screenshot(s)**
No screenshots, seg fault closes the GUI during STL loading

**Actual results**
Cura will start attempting to load the STL, then seg fault.

**Expected results**
The STL should load normally and Cura should not seg fault.

**Project file**
A couple of larger STLs that have caused this:

https://www.thingiverse.com/thing:4062723/files
https://www.thingiverse.com/thing:4423581/files
https://www.thingiverse.com/thing:4038181/files (Baby_Yoda_v2.2.stl)

**Log file**
```
2020-09-09 16:24:34,344 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [214]: Error check finished, result = False, time = 0.0s
2020-09-09 16:24:38,717 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [155]: New error check scheduled.
2020-09-09 16:24:38,718 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [214]: Error check finished, result = False, time = 0.0s
Empty filename passed to function
2020-09-09 16:24:48,641 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2020-09-09 16:24:54,955 - INFO - [Thread-3] STLReader.STLReader.load_file [60]: Using legacy code to load STL data.
2020-09-09 16:25:01,771 - DEBUG - [Thread-3] UM.Mesh.MeshData.calculateNormalsFromVertices [552]: Calculating normals took 0.07500791549682617 seconds
2020-09-09 16:25:01,786 - DEBUG - [Thread-3] STLReader.STLReader._read [104]: Loaded a mesh with 3000984 vertices
2020-09-09 16:25:02,525 - DEBUG - [Thread-3] UM.Mesh.MeshData.approximateConvexHull [505]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.7362217903137207 seconds. 1043 input vertices. 1043 output vertices.
Fatal Python error: Segmentation fault

Thread 0x00007f4e567fc700 (most recent call first):
  File "/usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "/usr/lib64/python3.8/threading.py", line 870 in run
  File "/usr/lib64/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/usr/lib64/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007f4e56ffd700 (most recent call first):
  File "/usr/lib64/python3.8/threading.py", line 306 in wait
  File "/usr/lib/python3.8/site-packages/zeroconf/__init__.py", line 2232 in wait
  File "/usr/lib/python3.8/site-packages/zeroconf/__init__.py", line 1539 in run
  File "/usr/lib64/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/usr/lib64/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007f4e577fe700 (most recent call first):
  File "/usr/lib64/python3.8/threading.py", line 306 in wait
  File "/usr/lib64/python3.8/threading.py", line 558 in wait
  File "/usr/lib/cura/plugins/UM3NetworkPrinting/src/Network/ZeroConfClient.py", line 81 in _handleOnServiceChangedRequests
  File "/usr/lib64/python3.8/threading.py", line 870 in run
  File "/usr/lib64/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/usr/lib64/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007f4e57fff700 (most recent call first):
  File "/usr/lib64/python3.8/threading.py", line 306 in wait
  File "/usr/lib/python3.8/site-packages/zeroconf/__init__.py", line 2232 in wait
  File "/usr/lib/python3.8/site-packages/zeroconf/__init__.py", line 1317 in run
  File "/usr/lib64/python3.8/threading.py", line 932 in _bootstrap_inner
[... 141 lines cut ...]
  File "/usr/lib64/python3.8/threading.py", line 302 in wait
  File "/usr/lib64/python3.8/threading.py", line 433 in acquire
  File "/usr/lib/python3.8/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.8/site-packages/UM/JobQueue.py", line 123 in run
  File "/usr/lib64/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/usr/lib64/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007f4ed0cba700 (most recent call first):
  File "/usr/lib64/python3.8/threading.py", line 302 in wait
  File "/usr/lib64/python3.8/threading.py", line 433 in acquire
  File "/usr/lib/python3.8/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.8/site-packages/UM/JobQueue.py", line 123 in run
  File "/usr/lib64/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/usr/lib64/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007f4effc13740 (most recent call first):
  File "/usr/lib/python3.8/site-packages/UM/Qt/QtApplication.py", line 378 in exec_
  File "/usr/lib/python3.8/site-packages/cura/CuraApplication.py", line 844 in run
  File "/usr/bin/cura", line 234 in <module>
Segmentation fault

```

**Additional information**
The same STLs load fine in both Luban and Simplify 3D

nallath, 2020-09-10:
That's odd. I've tried loading the baby Yoda file (although i am running it from source) and I didn't get a crash. Does it happen every time that you load a large file?

eoyilmaz, 2020-09-10:
I have the same problem with every STL file I try to load ((big or small) I get a crash. I'm on Fedora 32 too. Here is my outputs:

Error Traceback:
```
Traceback (most recent call last):
  File "/build/4.7/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 1060, in event
  File "/build/4.7/build/inst/lib/python3.5/site-packages/UM/Qt/QtApplication.py", line 464, in event
  File "/build/4.7/build/inst/lib/python3.5/site-packages/UM/Event.py", line 218, in call
  File "/build/4.7/build/inst/lib/python3.5/site-packages/UM/Signal.py", line 332, in __performEmit
  File "/build/4.7/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 1820, in _readMeshFinished
IndexError: list index out of range
```

Logs:
```
Thread 0x00007fae717fa700 (most recent call first):
  File "/tmp/.mount_Ultima6SJUvP/usr/bin/plugins/plugins/RemovableDriveOutputDevice/RemovableDrivePlugin.py", line 61 in _updateThread
  File "/build/env/4.7/inst/lib/python3.5/threading.py", line 862 in run
  File "/build/env/4.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007fae71ffb700 (most recent call first):
  File "/build/env/4.7/inst/lib/python3.5/threading.py", line 297 in wait
  File "/build/env/4.7/inst/lib/python3.5/site-packages/zeroconf/__init__.py", line 2117 in wait
  File "/build/env/4.7/inst/lib/python3.5/site-packages/zeroconf/__init__.py", line 1434 in run
  File "/build/env/4.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007fae727fc700 (most recent call first):
  File "/build/env/4.7/inst/lib/python3.5/threading.py", line 297 in wait
  File "/build/env/4.7/inst/lib/python3.5/threading.py", line 549 in wait
  File "/tmp/.mount_Ultima6SJUvP/usr/bin/plugins/plugins/UM3NetworkPrinting/src/Network/ZeroConfClient.py", line 81 in _handleOnServiceChangedRequests
  File "/build/env/4.7/inst/lib/python3.5/threading.py", line 862 in run
  File "/build/env/4.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007fae72ffd700 (most recent call first):
  File "/build/env/4.7/inst/lib/python3.5/threading.py", line 297 in wait
  File "/build/env/4.7/inst/lib/python3.5/site-packages/zeroconf/__init__.py", line 2117 in wait
  File "/build/env/4.7/inst/lib/python3.5/site-packages/zeroconf/__init__.py", line 1262 in run
  File "/build/env/4.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007fae737fe700 (most recent call first):
  File "/build/env/4.7/inst/lib/python3.5/site-packages/zeroconf/__init__.py", line 1183 in run
  File "/build/env/4.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007fae73fff700 (most recent call first):
  File "/tmp/.mount_Ultima6SJUvP/usr/bin/plugins/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "/build/env/4.7/inst/lib/python3.5/threading.py", line 862 in run
  File "/build/env/4.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007fae7b7fe700 (most recent call first):
  File "/build/4.7/build/inst/lib/python3.5/site-packages/UM/Backend/Backend.py", line 165 in _storeStderrToLogThread
  File "/build/env/4.7/inst/lib/python3.5/threading.py", line 862 in run
  File "/build/env/4.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007fae7bfff700 (most recent call first):
  File "/build/4.7/build/inst/lib/python3.5/site-packages/UM/Backend/Backend.py", line 153 in _storeOutputToLogThread
  File "/build/env/4.7/inst/lib/python3.5/threading.py", line 862 in run
  File "/build/env/4.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007fae997fa700 (most recent call first):
  File "/build/env/4.7/inst/lib/python3.5/threading.py", line 293 in wait
  File "/build/env/4.7/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/build/4.7/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 98 in _nextJob
[... 124 lines cut ...]
Current thread 0x00007faefdc6a740 (most recent call first):
  File "/build/4.7/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py", line 393 in _logInfoWidget
  File "/build/4.7/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py", line 182 in _createDialog
  File "/build/4.7/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py", line 86 in __init__
  File "/build/4.7/build/inst/bin/cura", line 166 in exceptHook
  File "/build/4.7/build/inst/lib/python3.5/site-packages/UM/Qt/QtApplication.py", line 378 in exec_
  File "/build/4.7/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 844 in run
  File "/build/4.7/build/inst/bin/cura", line 234 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
[... cut ...]
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
  File "/build/env/4.7/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
```

And here is the STL file that I try to open: https://www.dropbox.com/t/OYY0zNx0iXx26TwF

TurboTurtle, 2020-09-10:
> That's odd. I've tried loading the baby Yoda file (although i am running it from source) and I didn't get a crash. Does it happen every time that you load a large file?

Yep, every time.

Ghostkeeper, 2020-09-11:
For reference, the traceback is pointing to this line:

https://github.com/Ultimaker/Cura/blob/d115efbbb260bb6ae627273eff6fb322af51ac98/cura/CuraApplication.py#L1820

Ghostkeeper, 2020-09-11:
Actually I think we're looking at https://github.com/Ultimaker/Cura/issues/8235 . The crash traceback is exactly the same.

In 4.7.0 there was an issue with settings that contained an `=` symbol in the value, which was sometimes in people's start g-code. If your start g-code contained an `=` symbol, it would corrupt that file in memory. This caused the crash in #8235. For most people this got fixed in 4.7.1 because the file is properly read now. However if you then modified the printer while it was corrupt, the corruption would get saved to the file and be permanent.

A workaround for you is to create a new printer and try it there. Delete the old printer. You'll have to rewrite the start g-code though, since that is permanently lost.

Ghostkeeper, 2020-09-11:
Sorry, I didn't see that Eoyilmaz was not the same as TurboTurtle. His crash is unrelated to this issue.

casiosmu, 2020-09-12:
I can confirm this under OpenSuse Tumbleweed. The debug log is exactly the same.
For me the critical STL size is between 7 and 9 MB. A 7MB file opens correctly, with a 9MB (https://www.thingiverse.com/thing:3867211) or bigger file it shows the exact same behaviour as described by the issue opener.

eoyilmaz, 2020-09-14:
Today I've updated 4.6.1 to 4.7.1 through Fedora 32 repositories and the same error continues. I think as a Python developer I should give you guys a hand for this bug :+1: 

```
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/cura/CuraApplication.py", line 1060, in event
    return super().event(event)
  File "/usr/lib/python3.8/site-packages/UM/Qt/QtApplication.py", line 464, in event
    event._function_event.call()
  File "/usr/lib/python3.8/site-packages/UM/Event.py", line 218, in call
    self._function(*self._args, **self._kwargs)
  File "/usr/lib/python3.8/site-packages/UM/Signal.py", line 332, in __performEmit
    func(dest, *args, **kwargs)
  File "/usr/lib/python3.8/site-packages/cura/CuraApplication.py", line 1820, in _readMeshFinished
    default_extruder_id = self._global_container_stack.extruderList[int(default_extruder_position)].getId()
IndexError: list index out of range
```

eoyilmaz, 2020-09-14:
Okay one new thing, removing the ``~/.local/share/cura/4.7`` folder solved the problem! @Ghostkeeper I think you were right about removing the printer :+1:

TurboTurtle, 2020-09-14:
FWIW, I removed `~/.local/share/cura/4.7` as well but the seg fault issue remains for me.

nallath, 2020-09-15:
The segfault is a different issue from the "list index out of range". The second issue can be solved by removing the 4.7 configuration.

Ghostkeeper, 2020-09-15:
So TurboTurtle's crash seems to be a segfault in Numpy on this line:

https://github.com/Ultimaker/Uranium/blob/3ed9c4de76509edfd87ebabd9d7905853e11cb36/UM/Mesh/MeshData.py#L400

We haven't been able to reproduce that yet though.

GregValiant, 2024-11-20:
Is this still an issue in current versions of Cura (5.8.0 and up)? Can this be closed?

TurboTurtle, 2024-11-20:
I had completely forgotten about this, but can confirm no longer an issue with recent Cura.

closed 2024-11-20 by GregValiant