Ultimaker/Cura #8360: [4.7] Modifiers do not apply to all objects

Zogar89, 2020-09-10:
**Application version**
4.6, 4.7

**Platform**
Windows 10

**Printer**
Custom Artillery x1

**Reproduction steps**
1. Use an object (such as a cube) to apply modifiers to a part of another object.
2. Group the objects and multiply them several times.
3. Slice

**Screenshot(s)**
[image]

**Actual results**
Some objects do not have the modification applied.

If you delete the modifier on the multiplied objects that doesn't work and recreate it one by one from scratch, the objects where the modifier was applied before will no longer be modified.

**Expected results**
All objects have the modification applied.

**Project file**
https://drive.google.com/file/d/1Y46WkG5p8cmvfUi6eifirV9_TbZn4XVK/view?usp=sharing

nallath, 2020-09-10:
The models on the left and the right have different settings for the per object settings. I think you might have copied the models and then changed the settings on one of them. If I delete the right & left model and duplicate the middle one again, I get the 3x the same model.

Zogar89, 2020-09-10:
> 
> 
> The models on the left and the right have different settings for the per object settings. I think you might have copied the models and then changed the settings on one of them. If I delete the right & left model and duplicate the middle one again, I get the 3x the same model.

It is true that it has different configurations, it must be because I was doing tests but the settings of the middle model do not apply. Notice that it has 3% infill and is not being applied.

Ghostkeeper, 2020-09-11:
Funny. I'm getting a different result:
[image]
The middle one indeed has different settings, but that's only the wall line count and the top/bottom thickness so that shouldn't make a difference here.

I think the 0% infill density may be messing with it here. Changing that to 0.01% fixes the issue. When the infill is 0% it skips the whole infill generation. Maybe it skips a bit too much.

Zogar89, 2020-09-11:
> I think the 0% infill density may be messing with it here. Changing that to 0.01% fixes the issue. When the infill is 0% it skips the whole infill generation. Maybe it skips a bit too much.

I appreciate your solution to be able to get out of the way.

But this is a bug and should be fixed. It's not the first time that happens to me

Ghostkeeper, 2020-09-15:
Yes, but it then does allow the infill meshes to work so it can be set to 0% again then.

closed 2024-12-08 by github-actions[bot]