Ultimaker/Cura #8361: Cura error when make in ubuntu:  cx_Freeze.freezer.ConfigError: cannot find file/directory named /lib/libprotobuf.so.3.9.2.0 failed

JaideepBgit, 2020-09-10:
--------------------------------------------------------------------------------------------------------
--------------------------------------------------------------------------------------------------------
**Application version**
The application is taken from the below URL:
https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source-on-Ubuntu
--------------------------------------------------------------------------------------------------------
--------------------------------------------------------------------------------------------------------
**Platform**
NAME="Ubuntu"
VERSION="20.04.1 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.1 LTS"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal

--------------------------------------------------------------------------------------------------------
--------------------------------------------------------------------------------------------------------
**Reproduction steps**
7. Clone the cura-build repository.
8. Create a build subdirectory in that repository.
9. Open CMake and point its source directory to your cura-build repository and the build directory to the build subdirectory.
10. Configure and generate. Use the GCC make files to compile these.
11. Open a terminal and run make in the build directory. If all goes well, this should build Cura's main components.

--------------------------------------------------------------------------------------------------------
--------------------------------------------------------------------------------------------------------
ERROR after following the below steps:
------------------------------------
[ 98%] Packaging Projects...
Cleaning old package directory...
Running cx_Freeze to generate executable...
running build
running build_exe
Traceback (most recent call last):
  File "setup.py", line 148, in <module>
    setup(
  File "/usr/local/lib/python3.8/dist-packages/cx_Freeze/dist.py", line 349, in setup
    distutils.core.setup(**attrs)
  File "/usr/lib/python--------------------------------------------------------------------------------------------------------
--------------------------------------------------------------------------------------------------------
**Application version**3.8/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/usr/lib/python3.8/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python3.8/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/usr/lib/python3.8/distutils/command/build.py", line 135, in run
    self.run_command(cmd_name)
  File "/usr/lib/python3.8/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.8/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/usr/local/lib/python3.8/dist-packages/cx_Freeze/dist.py", line 203, in run
    freezer = cx_Freeze.Freezer(self.distribution.executables,
  File "/usr/local/lib/python3.8/dist-packages/cx_Freeze/freezer.py", line 149, in __init__
    self._VerifyConfiguration()
  File "/usr/local/lib/python3.8/dist-packages/cx_Freeze/freezer.py", line 467, in _VerifyConfiguration
    raise ConfigError("cannot find file/directory named %s",
cx_Freeze.freezer.ConfigError: cannot find file/directory named /lib/libprotobuf.so.3.9.2.0
make[2]: *** [CMakeFiles/packaging.dir/build.make:64: packaging] Error 1
make[1]: *** [CMakeFiles/Makefile2:99: CMakeFiles/packaging.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

Expected results
Should build all dependencies successfully

Ghostkeeper, 2020-09-11:
I can't reproduce this issue.

Cura is normally built using Python 3.5. Maybe you need to downgrade to that. It's also probably a good idea to install things to a custom directory as to not mix up Cura's dependencies with the packages on your system. I can understand that it might take the wrong version of Protobuf otherwise.
Is there a libprotobuf in that folder?

JaideepBgit, 2020-09-14:
@Ghostkeeper When using Python3.5 there were many errors, but I understand now that those errors can now be overlooked and libraries can be separately installed and proceed with the build process.

closed 2020-09-14 by JaideepBgit