Ultimaker/Cura #8367: [4.7.1] Crash at Profile Selection first run

gdampf, 2020-09-11:
**Application version**
4.7.1

**Platform**
Windows 10 pro 64bit

**Printer**
Crealitiy Ender 3

**Reproduction steps**
1. Upgrade from 4.6.2 to 4.7.1
2. Step thru first run questions
3. Select unconnected Printer Profile: Creality Ender 3
4. it crashes
5. restart it
6. Step thru first run questions again
7. immediately after the question to allow anonymous data, it crashes

**Screenshot(s)**
Last step
[image]

**Actual results**
Crash

**Expected results**
Don't crash

**Project file**
None

**Log file**
2020-09-11 10:12:10,841 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin ConsoleLogger
2020-09-11 10:12:10,856 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin CuraDrive
2020-09-11 10:12:10,894 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.__init__ [92]: Found CuraEngine at: C:\Program Files\Ultimaker Cura 4.7\CuraEngine.exe
2020-09-11 10:12:10,910 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin CuraEngineBackend
2020-09-11 10:12:10,925 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin CuraProfileReader
2020-09-11 10:12:10,925 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [188]: Socket state changed to Listening
2020-09-11 10:12:10,925 - INFO - [MainThread] UM.Backend.Backend.startEngine [90]: Started engine process: C:\Program Files\Ultimaker Cura 4.7\CuraEngine.exe
2020-09-11 10:12:10,941 - DEBUG - [MainThread] UM.Backend.Backend._backendLog [110]: [Backend] Calling engine with: ['C:\\Program Files\\Ultimaker Cura 4.7\\CuraEngine.exe', 'connect', '127.0.0.1:49674', '']
2020-09-11 10:12:10,957 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin CuraProfileWriter
2020-09-11 10:12:10,972 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin OBJReader
2020-09-11 10:12:10,994 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin OBJWriter
2020-09-11 10:12:10,994 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2020-09-11 10:12:10,994 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [192]: Socket state changed to Connected
2020-09-11 10:12:10,994 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] Cura_SteamEngine version 4.7.1
2020-09-11 10:12:11,010 - DEBUG - [MainThread] UM.Backend.Backend._onSocketStateChanged [181]: Backend connected on port 49674
2020-09-11 10:12:11,010 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] Copyright (C) 2020 Ultimaker
2020-09-11 10:12:11,010 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2020-09-11 10:12:11,010 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] This program is free software: you can redistribute it and/or modify
2020-09-11 10:12:11,026 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] it under the terms of the GNU Affero General Public License as published by
2020-09-11 10:12:11,026 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] the Free Software Foundation, either version 3 of the License, or
2020-09-11 10:12:11,026 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] (at your option) any later version.
2020-09-11 10:12:11,026 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2020-09-11 10:12:11,041 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] This program is distributed in the hope that it will be useful,
2020-09-11 10:12:11,041 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] but WITHOUT ANY WARRANTY; without even the implied warranty of
2020-09-11 10:12:11,041 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
2020-09-11 10:12:11,041 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] GNU Affero General Public License for more details.
2020-09-11 10:12:11,041 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] 
2020-09-11 10:12:11,057 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] You should have received a copy of the GNU Affero General Public License
2020-09-11 10:12:11,057 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] along with this program.  If not, see <http://www.gnu.org/licenses/>.
2020-09-11 10:12:11,072 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin STLReader
2020-09-11 10:12:11,095 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin STLWriter
2020-09-11 10:12:11,095 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FileLogger
2020-09-11 10:12:11,126 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FirmwareUpdateChecker
2020-09-11 10:12:11,141 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FirmwareUpdater
2020-09-11 10:12:11,157 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeGzReader
2020-09-11 10:12:11,157 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeGzWriter
2020-09-11 10:12:11,173 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeProfileReader
2020-09-11 10:12:11,195 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeReader
[... 206 lines cut ...]
Thread 0x00000f1c (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Current thread 0x00002610 (most recent call first):
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 378 in exec_
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 844 in run
  File "X:\4.7-exe\build\inst\bin\cura_app.py", line 234 in <module>
  File "Console.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
  File "__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load

**Additional information**
None

piroxxi, 2020-09-13:
Hi, 
I have the exact same issue.

konskarm, 2020-09-14:
Hi @gdampf. From a clean install of 4.7 I am unable to reproduce this. The Creality Ender 3-Pro profile loads whitout issues for me.
When you were uninstalling 4.6.2 did you keep your configurations?
Could you try deleting 4.6 and 4.7 configurations (assuming you won't lose your data) and then try loading Cura 4.7 again?

gdampf, 2020-09-14:
No, I selected to purge them. 
But, what I found is, that not all profiles were deleted this way. One added for DaVinci Printer, which wasn't included in the clean install, remained and caused this kind of trouble, I guess. Interesting is, that the Version 4.6.2 just ignored the faulty printer description, the 4.7.1 crashed instead.
Maybe this is the only issue. I didn't test it completely. I will keep the older version, which is more reliable for a while and may give 4.7.2 a chance, when released.

Ghostkeeper, 2020-09-15:
Hmm, the stack trace indicates pretty much nothing. The Qt GUI framework is getting a segfault, which should never happen unless there is a bug in Qt somewhere.

The window after the anonymous data collection screen is the log-in prompt. So it would appear there is something there that is causing the crash? Only, that screen is basically just an SVG image, some text, two buttons referring to a website when clicked, and a "skip" button. Nothing really special and pretty much a subset of the very first "welcome" screen.

If you purge your profiles, be sure to do that while all versions of Cura are closed. Otherwise it'll indeed re-save some of them when Cura is closed or you change a profile.

ShuzzDE, 2020-09-25:
Hi!
I'm getting the exact same behaviour with a virtually identical logfile.
Funny thing is: The machine I'm getting the error on has been freshly installed from scratch.
Empty hard drive, fresh win10 install, fresh Cura download + install --> Crashes at the exact same point.

RickUsingGitHub, 2020-10-04:
Same deal. Cleaned out all old versions emptying profiles from %appdata%Cura.

Crashes sometimes at that OP location sometimes next page after that.
Subset of my log.
`2020-10-04 13:26:23,683 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2020-10-04 13:26:31,487 - INFO - [MainThread] cura.CuraApplication.writeToLog [595]: User accepted the User-Agreement.
Fatal Python error: Segmentation fault

Thread 0x00000e98 (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.7\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 79 in _updateThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00002e70 (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.7\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 61 in _updateThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00001178 (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf\__init__.py", line 2117 in wait
  File "zeroconf\__init__.py", line 1434 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x0000203c (most recent call first):
  File "threading.py", line 297 in wait
  File "threading.py", line 549 in wait
  File "C:\Program Files\Ultimaker Cura 4.7\plugins\UM3NetworkPrinting\src\Network\ZeroConfClient.py", line 81 in _handleOnServiceChangedRequests
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000033a4 (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf\__init__.py", line 2117 in wait
  File "zeroconf\__init__.py", line 1262 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00001528 (most recent call first):
  File "zeroconf\__init__.py", line 1183 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000017c4 (most recent call first):
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\Backend\Backend.py", line 165 in _storeStderrToLogThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00001eb8 (most recent call first):
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\Backend\Backend.py", line 153 in _storeOutputToLogThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00000cb4 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00001ee8 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x0000269c (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00001dc4 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Current thread 0x00000b7c (most recent call first):
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 378 in exec_
  File "X:\4.7-exe\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 844 in run
  File "X:\4.7-exe\build\inst\bin\cura_app.py", line 234 in <module>
  File "Console.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
  File "__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load`

awb95, 2020-10-17:
Hi,

I'm having the same problem, but running Cura with my second Graphic Card instead the integrated one solved the problem for me.

doc-Seltsam, 2020-10-22:
Hi,

I have the exact same issue, with Cura 4.70 and also 4.71

Win10 64Bit

"Fatal Python error: Segmentation fault"

during initial startup after "help us improving .." screen and before the screen to choose a printer appears.

Cura 4.61 works fine, before the update and after roll back everything.

I tried al lot, cleaning %appdata%, cleaning registry, paused my pi-hole, uninstalled driver, switched off network, ...

But I'm happy with 4.61 so far. I had an backup of my materials and profiles.

doc-Seltsam, 2020-11-28:
More or less by chance I found a solution to the problem.  Maybe it is just a solution of my problem!

The crash is apparently caused by a problem with my OpenGL driver for the internal Intel HD3000 graphics. The Intel drivers are too old.
To fix this, install the "Application Compatibility Tools" from the Microsoft Windows ADK package.

In this tool you give Cura the compatibility fix WIN81RTMVersionLie.
Then you start Cura the first time from this tool. This will run it through the initialization. The next times you start Cura as normal.

[image]

3dmakertech1, 2020-11-29:
Outstanding.
Works just fine now.
Thanks Doc!

cyberws, 2020-12-05:
For me this issue was fixed quite easily (given the hints above at to the GPU issue)

I have a laptop with an Intel HD3000 on motherboard and additional 2GB Ram Nvidia GeForce GPU.

Navigate to the location of the Application exe file, most likely C:\Program Files\Ultimaker Cura 4.8.0

When I right click on the Application there is a Run with Graphic Process pop-out menu. Once I have run the program once using the "High Performance NVIDIA processor" option. It seems to run thereafter without issue (not tested following a reboot at this time).

[image]

Hope this helps.

Regards,

Phil - I don't have 3D Printer yet, Ender 3 Pro on order, but getting up to speed before it arrives.

tonyctp, 2021-07-18:
I had occupied the same issues. It works fine in my linux debian 11 AppImage but win10 workstation failed.
After several test, it works fine under my administrator account in win10, it may be caused by specisal account name (I had used chinese as my user name)
I hope it will be helpful for you
Logs as below, the path is invalid not existed in my disk.
2021-07-18 23:41:29,728 - ERROR - [JobQueueWorker [2]] UM.Logger.logException [116]:   File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-exe\build\inst\lib\python3.8\site-packages\UM\Mesh\MeshData.py", line 220, in getExtents
2021-07-18 23:41:29,733 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [66]: An uncaught error has occurred!

closed 2024-12-08 by github-actions[bot]