Ultimaker/Cura #8368: [4.3.0] Change putValue() helper function to accept unnamed args

iliya-malecki, 2020-09-11:
I would like to use M117 command in my post-processing plug-in and still use all the convenience functions, namely, putValue().
However, it supports only **kwargs, so the code ends up being quite disgusting (see example). Also, if i needed to add messages that start with letter M or G that could mess up the whole command priority, on top of being funny yet disgustingly awful

**Application version** 
4.3.0

**Platform** 
Windows 10

**Actual results**

` layerCommandLines.insert(command_idx + 1, self.putValue("", M=117, P="rinting helpers"))`
`layerCommandLines.insert(command_idx + 1, self.putValue("", M=486, S=-1))`

**Expected results**

`layerCommandLines.insert(command_idx + 1, self.putValue("", M=117, "Printing helpers"))`
 `layerCommandLines.insert(command_idx + 1, self.putValue("", M=486, S=-1))`
or
`layerCommandLines.insert(command_idx + 1, self.putValue("M117, "Printing helpers"))`
 `layerCommandLines.insert(command_idx + 1, self.putValue("", M=486, S=-1))`

nallath, 2020-09-11:
> `layerCommandLines.insert(command_idx + 1, self.putValue("", M=117, "Printing helpers"))`

This is invalid python. A kwarg argument can not be given before an arg. 

But why not just use:
`layerCommandLines.insert(command_idx+1, "M117") + "printing helpers")`

iliya-malecki, 2020-09-13:
wow okay, that really does make a lot of sense, why would i ever put args after kwargs?

closed 2020-09-13 by iliya-malecki