Ultimaker/Cura #8371: When I update the price for one custom filament it doesn't seem to save.  When I switch to another filament, the price stays the same as previously selected filament.

DavidRizzo, 2020-09-11:
**Application version**
Ultimaker Cura Version 4.7.1

**Platform**
Windows 10
Processor: Intel(R) Core (TM) i7-7700 CPU @ 3.60GHz 
Ram: 16 GB
System Type: 64-Bit OS, Windows

**Printer**
Creality Ender-3 Pro

**Reproduction steps**
1. Preferences **Configure Cura**
2. Manage **Materials**
3. Entered **Filament Cost**
4. Switched to another custom filament and the price did not change
5. One Price was $22.99 the other was $23.99, when I switched from the one that costs $23.99 to the one that costs $22.99 the price stayed $23.99.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
**This one should be $23.99**
[image]

**This one should be $22.99**
[image]

**Actual results**
When I switched from the one that costs $23.99 to the one that costs $22.99 the price stayed $23.99.

**Expected results**
The price should change to $22.99 and not remain $23.99.

- Strongman is $23.99
- Hatchbox is $22.99

**Project file**
N/A

**Log file**
N/A

**Additional information**
I'm using this to track materials, it would be nice if I could also clear out materials I am not using and only see my inventory.
The important additional information is that each field be SAVED this information is important!

fieldOfView, 2020-09-12:
See the remark about the materials being "linked" in your screenshot. Linked materials share price information. If you "unlink" them you can enter a different price per material.

closed 2020-09-15 by nallath

DavidRizzo, 2020-09-15:
That resolved it, thanks.  Me happy again!

Ghostkeeper, 2020-09-17:
The UX of this is less than ideal. It's communicated through a message saying "some metadata is shared", but it's not really intuitive or clear to the user.