Ultimaker/Cura #8372: jittering/slowdown when X and Y move at same time. causes rough walls

Noinimm, 2020-09-12:
**Application version**
4.7.1, but i believe it started on 4.7 before update

**Platform**
windows 10

**Printer**
ender 3

**Reproduction steps**
1. updated cura to most recent version
2. printed a benchy

**Screenshot(s)**
[image]

**Actual results**
when printing in a straight line, there is no issue, but once it gets to a curve where X and Y are both moving, it slows down and slightly jitters while traveling

**Expected results**
very rough walls that could be mistaken for overextrusion or some other issue (and drive a man mad for a week of troubleshooting and large number of canceled benchys

**Project file**
[CE3_3DBenchy.zip](https://github.com/Ultimaker/Cura/files/5212041/CE3_3DBenchy.zip)

**Log file**
not a software crash, let me know if this is still needed

**Additional information**

this issue disappeared when i downgraded to cura 4.6.2

i have an early model of ender 3 (prior to the pro being released).
board does not have silent drivers.
i am running a th3d ezabl
the gcode was sent from octoprint 1.4.2

my starting gcode, possibly something in here causing the interaction?:
M140 S{material_bed_temperature_layer_0} ; Set Bed Temp
M190 S{material_bed_temperature_layer_0} ; Wait for Bed Temp
G28 ; Home all axes
M104 S{material_print_temperature_layer_0} ; Set Extruder Temp
@BEDLEVELVISUALIZER ;
G29 ; Auto Bed Leveling
G92 E0 ; Reset Extruder
G1 Z10.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z10.0 F4000.0 ; Move to pre-start position
G1 X0.1 Y20 Z0.3 F4000.0 ; Move to start position
M109 S{material_print_temperature_layer_0} ; Wait for Extruder Temp
G1 X0.1 Y200.0 Z0.3 F1500.0
E15 ; Draw the first line
G1 X0.5 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.5 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z3.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed

Noinimm, 2020-09-12:
did not notice the pinned issue before posting this.
i'm just relieved it isn't just me

lilasiian8, 2020-09-12:
[image]

I had similar results and issues with 4.7/4.7.1. Echoing above, I thought it was just me... or my Klipper e-Steps setting!

ils15, 2020-09-12:
same here
[image]

Zogar89, 2020-09-12:
Duplicate of  8321

konskarm, 2020-09-14:
Hi @Noinimm. We are currently working on addressing this issue. 
As @Zogar89 mentions, this is a duplicate of https://github.com/Ultimaker/Cura/issues/8321, so I am closing this one. Please move further comments to the original issue.

closed 2020-09-14 by konskarm