Ultimaker/Cura #8373: Post Processing Script "Pause at height" does not work with 4.7.1 at all

LairdPaul, 2020-09-13:
I updated to 4.7.1 and tried a Pause at height test print again.  It never paused at height just kept on printing.  So I opened my Gcode file and searched for "M0, Pause, Change-Color" and they came up as not found.  I re-sliced and searched again with the same results??  I can only conclude the Pause at height Script is not working.

I re-sliced with the change filament script and searched my Gcode file for M600 and found the line --M600 E30.00 L50.00 X10.00 Y10.00 ; Generated by FilamentChange plugin. I printed and it worked with this script ok except a blob  when it changed Filament.
[image]

LairdPaul, 2020-09-13:
Oops forgot I'm using the Marlin M0 script on a Creality CR10 S4.
Thanks

konskarm, 2020-09-14:
I am guessing this isn't an issue anymore, so I am closing it.

closed 2020-09-14 by konskarm

LairdPaul, 2020-09-14:
It hasn't been fixed that I know of???   I even tried uninstalling Cura and reinstalling Cura about 4 hours ago with the same result. and I haven't seen and updates???

LairdPaul, 2020-09-14:
I've been working on this all Night and Day.  I did try the old script from both 4.6 and 4.5 and  got the same results. So I uninstalled Cura deleted and folders or files left behind then reinstalled it,  still not working. I went to my laptop installed 4.5, Same results.  Went to an old PC with Cura 3.6 it worked!!  I went back to my main PC and reinstalled it over the top of the 4.7.1 install,  and WOW all is working now.  Don't know what fixed it but it works.

funx24x7, 2020-10-02:
M0 worked for me on my stock Ender 3 Pro, after upgrading to a silent motherboard it experienced similar issues with M0 pauses resuming automatically without user input. After some tinkering I found that adding the expire time parameter (ex. M0 S120) made the pause work properly.