Ultimaker/Cura #8374: lost all my profiles after install of this 4.7.1 coming from 4.7.0

knucklesdamone, 2020-09-13:
**Application version**
(The version of the application this issue occurs with.)
4.7.1
**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
windows 10 latest stable updates not preview version just normal win10 x64 with 1070 ti
**Printer**
(Which printer was selected in Cura?)
emder 3 pro
**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)
uninstalled 4.7.0 as installed asked and i clicked no to remove profile and then it installed new version left all as default and I have no profiles but i have all my printers setup my materials are fine but profiles are  gone now
**Screenshot(s)**
https://imgur.com/a/OvhensY
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)
no profiles show up
**Expected results**
(What should happen after the above steps have been followed.)
profile always show up after upgrades
**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

petemagnusson, 2020-09-14:
Same for me.
Uninstall and removed the config folder. And followed all the steps.
What i need to do now is to uninstall, remove files and reinstall and add profiles then i can print. Closing the program however forces me to redo all these steps.

knucklesdamone, 2020-09-14:
I did the same thing I uninstalled reinstalled copied my profile from the
old version 4.6 to the new one and my profiles are back

nallath, 2020-09-15:
Duplicate of https://github.com/Ultimaker/Cura/issues/8229

closed 2020-09-15 by nallath