Ultimaker/Cura #8375: nozzle settings not change

m0alves, 2020-09-13:
**Application version**
4.7.1

**Platform**
Windows 10 version 1909 OS Build 18363.1016

**Printer**
Anet A8

**Reproduction steps**
1. manage printers | Machine settings | Extruder 1 tab | Nozzle size = 0.6 mm (it shows the default 0.4, and I change it)
 
**Screenshot(s)**
[image]

**Actual results**
After "close" (on machine settings) and Close on Printers, if I go back to the Nozzle size settings (same way), it shows default 0.4 instead 0.6 I entered.

**Expected results**
nozzle size settings it's not save / it doesn't change. Whatever the value I enter, it´s always default value 0.4 that is assumed.

**Project file**
NA

**Log file**

**Additional information**

rburema, 2020-09-15:
We've had problems in the past with different locales. Is your locale set to ',' or '.' as the decimal separator?

m0alves, 2020-09-15:
Hello Remco
thanks for your message.
regarding the issue, I tried with both '.' and ',' with the same result.
Thank you
M Alves

Ghostkeeper, 2020-09-17:
I'm unable to reproduce this issue. It stays properly at 0.6 then.

m0alves, 2020-09-17:
thank you anyway.
I´ll try another version... older versions that I used, didn't have this
issue.

Ghostkeeper, 2020-09-18:
What is your system's locale? Like, is it set to Dutch, French, etc?

m0alves, 2020-09-18:
Region formats are "Portuguese (Portugal)"

control panel | Region | Country or region, it's "Portugal"
control panel | Region | Regional Format, it's "Portuguese (Portugal)"

control panel | Language it´s "English (United States)"

(keyboard it's Portuguese)

please note that I tried with "." (as it was) and witn "," with the same
result (new value not saved).
  Thank you

m0alves, 2020-09-18:
Hi Ghostkeeper
you figured it out... it has to do with decimal separator

The nozzle size is presented with "." decimal separator (0.4).
if I just change change the last number (0.6), it doesn't save (because
"my" decimal separator should be ",")
if I use the "," separator (0,6), the number becomes "corrected" (0.6) and
doesn't work to
I found a way to enter the number (with "," separator) without giving the
chance for corrections - simply pressing "close" directly and still in edit
"mode" of the number.
It works as expected.

I suggest more feedback when the numbers entered are not accepted and
simply accept both decimal separators ( and "correct" / "show" then in the
format you like best).

Thank you
M Alves

johansenhr, 2020-09-19:
I have the same problem.

First: There is no field for changing nozzle size in the default Creality Ender 3 printer profile. I have to create a custom profile.

Second: I then somehow manage to type in 0.6mm nozzle size, and when I switch back and forth between Printer and Extruder tab, the value seems to be stored. But when I activate another printer profile and then switch back, the value has changed back to the default setting (0,4mm)

Does someone know how to bypass this and properly tell the printer I've installed a 0.6mm nozzle?

Ghostkeeper, 2020-09-21:
You could try to install the Printer Settings plug-in from the Marketplace, Johansenhr. The nozzle size is just listed among the main settings list then.

> First: There is no field for changing nozzle size in the default Creality Ender 3 printer profile. I have to create a custom profile.

This is intended. The Creality Ender 3 has profiles for nozzles from which you can select. These profiles override the nozzle size so changing the machine is not effective anyway then.

> Second: I then somehow manage to type in 0.6mm nozzle size, and when I switch back and forth between Printer and Extruder tab, the value seems to be stored. But when I activate another printer profile and then switch back, the value has changed back to the default setting (0,4mm)

Does the workaround from m0alves work for you?

closed 2020-12-15 by no-response[bot]