Ultimaker/Cura #8377: Post Processing Script "Pause at height (BQ printers)" in 4.7.1 drives printer crazy

Takedowns, 2020-09-13:
This script incorrectly pauses printing. If you pause manually, the printer lowers the print bed a certain distance, then moves the extruder to the edge of the table. When using this script, the print bed just lowers without moving the extruder, but the printer thinks that the extruder has also been moved to the edge of the table. Accordingly, after continuing to print, the printer is completely lost in space, almost breaking itself. This was the first time I had to use an emergency stop because the printer nearly broke all of its axes.

rburema, 2020-09-15:
Does it work with the 'normal' pause at height script?

Just a note: For 3rd party printers, we rely on the community for creating PR's and testing. If you know what should be different in the current script, please let us know.

Ghostkeeper, 2020-09-15:
> but the printer thinks that the extruder has also been moved to the edge of the table.

Why would the printer think that?

Takedowns, 2020-09-15:
> > but the printer thinks that the extruder has also been moved to the edge of the table.
> 
> Why would the printer think that?

Idk, maybe the script and the printer have different concept of pause. Also i think that maybe this script was made for the old firmware.

Ghostkeeper, 2020-09-17:
To fix this, we'll need to know what g-code commands your new firmware needs to have in order to properly pause then.

closed 2020-10-01 by no-response[bot]