Ultimaker/Cura #8378: [4.7.1] Print Setting "Tips" Remain After Print Setting Dialog Dismissed

mbmast, 2020-09-13:
[image]

**Application version**
4.7.1

**Platform**
Windows 10

**Printer**
Most likely, not applicable to this issue, but it was a custom printer setting (CreateBot 3D).

**Reproduction steps**
All I did was open the Print setting, reviewed a few settings and dismissed the dialog box.  I don't think I changed anythingl

**Screenshot(s)**
Attached

**Actual results**
The Tips dialog box ("Generate Support / Generate structures to support parts of the model which ...") remains orphaned on the screen, obscuring things behind it.

**Expected results**
When the Print dialog box is dismissed, I would expect any displayed Tips windows to also be dismissed.

**Project file**
Not included.  Cura functions fine.  No issues.  The only issue is the Tips window remains.

**Log file**
Attached, but probably not relevant as no crashes occured.

**Additional information**
(Extra information relevant to the issue.)

konskarm, 2020-09-14:
Hi @mbmast. I tried reproducing the issue, but once I move my cursor away from the setting, the tooltip disappears. Can you reproduce it reliably? Maybe this was a one time thing.

mbmast, 2020-09-14:
I can't reproduce it either.

closed 2020-09-14 by mbmast

Ghostkeeper, 2020-09-15:
This happens very rarely, but once in a while. We haven't been able to fix it since it's so rare, and the workaround of just hovering over any setting is easy enough.