Ultimaker/Cura #8380: [4.7] Mesh Type: Modify settings for overlay - Cutting Mesh does not remove bed adhesion/no setting

donkeykong3x3, 2020-09-14:
**Application version**
Cura 4.7

**Platform**
Windows 10

**Printer**
Tevo Tarantula Pro

**Reproduction steps**
1. Use Mesh Type: Modify settings for overlay to cut part.
2. Set wall, top, and bottom counts to 0
3. Set infill density to 0%

**Screenshot(s)**
[image]

**Actual results**
Bed adhesion is not removed, nor is there a setting option to disable.

**Expected results**
Preferably bed adhesion would be based on the resulting mesh being sliced.

Ghostkeeper, 2020-09-15:
Bed adhesion is a global setting at the moment. It can't be determined for each object separately. The reason for this is that the raft adds extra layers, and the brim and skirt might intersect with other objects then.

Ghostkeeper, 2020-09-15:
Closing this as duplicate of https://github.com/Ultimaker/Cura/issues/6882.

closed 2020-09-15 by Ghostkeeper