Ultimaker/Cura #8382: Change AtoZ no display apply layer

lipk888, 2020-09-14:
**Application version:(cura 4.7.)
**Platform：(WIN8 SURFACE PRO3.)
**Printer: (DIY UM2)
issue descripsion:
when i add "change A to Z plugins" and i want to apply muti-layer change e.g. layer No.1 to No.19,  then i choice apply to "target layer+subsequent layer "but it not show subsequent layer which i can set.
not sure if it is bugs or other issue?

[image]

rburema, 2020-09-15:
Subsequent layers keeps the change for _all_ subsequent layers rather then reverting the change for the next (as opposed to 'single layer'). If you want to change multiple _specific_ layers, you'll have to add multiple ChangeAtZ scripts in the 'single layer' mode.

Ghostkeeper, 2020-09-16:
So for instance you'd add a ChangeAtZ script that changes at layer 1 and subsequent layers, the print speed to 150%.

Then you'd add another ChangeAtZ script that changes from layer 19 and subsequent layers, the print speed to 66.7%, resulting in the rest of the print being printed at 100% speed again.

closed 2020-09-16 by Ghostkeeper

lipk888, 2020-09-18:
thanks team, it seems work now