Ultimaker/Cura #8383: [4.7] Support is printed where it shouldn't

GMagician, 2020-09-14:
**Application version**
Cura 4.7

**Platform**
Win10

**Printer**
Custom

**Reproduction steps**
1. Open project
2. Slice it

**Screenshot(s)**
[image]

**Actual results**
Base support is printer on whole base

**Expected results**
An hole in support should be present

**Project file**
[Part Studio 1 - Part 2.zip](https://github.com/Ultimaker/Cura/files/5219977/Part.Studio.1.-.Part.2.zip)

nallath, 2020-09-15:
Change the value of "support merge distance" to zero. The round circle is simply too small and it is filled by the merge distance.

GMagician, 2020-09-15:
@nallath thanks for your answer but I can't find any "support merge distance" option
[image]

rburema, 2020-09-15:
@GMagician I think it's support _join_ distance.

GMagician, 2020-09-15:
@nallath & @rburema it doesn't work either
[image]

GMagician, 2020-09-15:
@nallath & @rburema I found why it's not working and how to let it works, but it's a bug to me.

Overriding value where you see in picture: **it doesnt' work**.
With such override I enabled supports in print settings and I wrote 0 there in "support join distance": **it doesn't work**
With override 0, print setting->support join distance 0 I re-disabled support enabled: **it works**!

I think that that's definetively a bug, even if not the original posted

rburema, 2020-09-15:
That is strange indeed. I'll at least reopen it, and then we'll rediscuss how much priority this should get.

Ghostkeeper, 2020-09-16:
We had an incongruence here in how the setting is displayed in the front-end and how the setting behaves when slicing. Basically, the problem was that the front-end thinks it's a per-object setting but the back-end gets the setting value as a per-extruder setting (from the Support Infill Extruder).

Here's a workaround for you:
1. Enable support.
2. Set Support Join Distance to 0.
3. Disable support again.

It then shows the hole as expected.
[image]

I've fixed this for 4.8. The changes I made are:
* The setting can no longer be added as a per-object setting.
* The setting will show up now if support is disabled but there are support meshes in the scene (so you don't need to enable-change-disable support any more).

Same goes for Support Horizontal Expansion, which had the same problem.

commit 0029221f (2020-09-16): Set Support Join Distance and Support Horizontal Expansion as per-extruder settings

This is how CuraEngine's algorithms use them. They don't have access to which mesh the support originally belonged to.

Fixes #8383.
closed 2020-09-16 by Ghostkeeper