Ultimaker/Cura #8384: Ultimaker Cura 3.3 Linux ARM version crashes on RPI 4

zipCoder933, 2020-09-14:
I have installed Cura on my RPI 4 by using the `sudo apt-get install cura` terminal command. Whenever I try to run the app, however, i get this:

[![Cura running on RPI crashes][1]][1]

Something with the configuration files? I'm not sure. I have tried clicking the "Backup and reset configuration", and even deleting the configuration files folder. But it doesn't help.

Is there something I can do about this? Or is this just another Bug?

  [1]: https://i.stack.imgur.com/iKrZo.png

rburema, 2020-09-15:
Maybe it can't _read_ the configuration files? What are the file permissions set to?

The 3.3 version is quite old by now. We're on 4.7 by now (though there are some scenarios for which we recommend 4.6.2).

nallath, 2020-09-15:
We also don't support (or provide) Cura for ARM or via apt. The only one that we support is the Appimage. For any questions about that package, please contact the maintainer of the apt package.

closed 2020-09-15 by nallath