Ultimaker/Cura #8386: Printer profile "Rename" doesn't work

HBrydon, 2020-09-15:
**Application version**
Cura 4.7 and 4.7.1

**Platform**
Windows 10 Dell computer AMD Ryzen processor

**Printer**
Creality Ender-3 Pro

**Reproduction steps**
1. Start Cura. In my case, this was an upgrade from 4.6.x -> 4.7, followed a week later by 4.7 -> 4.7.1. In the previous version, I had already customized Creality Ender-3 Pro config in my settings. I discovered that my printer config list consisted of "Creality Ender-3 Pro" (stock) and "HB Mods to Creality Ender-3 Pro" (my customized version). The 4.6.x -> 4.7 upgrade botched bringing forward my customizations because I had an "=" sign in a comment in part of the customizations (hopefully 4.7.1 fixed this, haven't verified yet, not what this bug report is about).
2. Click "Manage Printers" either from the Preferences-> Configure Cura menu or from "Manage Printers" button on the printer list. Edit the "Creality Ender-3 Pro" config, making at least one change (pick one). Rename the "Creality Ender-3 Pro" config to something else, eg. "HB New Creality Ender-3 Pro". Everything looks good, click "close".
3. Re-enter the "Manage Printers" dialog. Everything still looks good, click "close".
4. Exit the program. Restart the program. Click "Manage Printers". Note that the renamed printer config is gone and the "Creality Ender-3 Pro" config is back.

**Screenshot(s)**
No images necessary.

**Actual results**
The "Rename" button only temporarily renames the config. Exit the program, restart, the rename is undone.

**Expected results**
The "Rename" button should rename the config, and the rename operation should survive exiting the program.

**Project file**
N/A

**Log file**
N/A

**Additional information**
N/A

Ghostkeeper, 2020-09-16:
I could reproduce this also from a fresh installation. My reproduction steps were:

1. Add Ultimaker 2+.
2. In the printer management screen, press the "Rename" button to rename the Ultimaker 2+ to "Renamed Ultimaker 2+".
3. Restart Cura.

The printer is just called "Ultimaker 2+" again.

nallath, 2020-09-18:
I've already done a bit of analysis on this. It seems that the renaming doesn't mark the printer as "dirty" (eg; it needs to be saved to file). 

If you for instance change a profile / material, the name is stored correctly.

The bug is likely to be picked up for 4.8

Vandrasc, 2020-09-18:
devs see Cura-7725 for our internal reference

closed 2020-09-22 by nallath