Ultimaker/Cura #8387: [4.7.1] Tree support option not visible

yael-lorenzo, 2020-09-15:
**Application version**
4.7.1

**Platform**
MacOs Catalina 10.15.6

**Printer**
Creality Ender 3 Pro

**Reproduction steps**
1. Open model
2. Go to slice options and search for Tree Support
3. The option is not there.
4. Go to visibility options and it is not there either

**Screenshot(s)**
[image]

**Actual results**
There are no tree support options.

**Expected results**
I should see the options

**Project file**
Any

**Additional information**
Cura has been behaving really strange since the last 3 updates.

yael-lorenzo, 2020-09-15:
F**k, I found the option under support structure, sorry for the noise.

closed 2020-09-15 by yael-lorenzo

DreadKnight, 2020-09-26:
Had issues as well finding it xD

Ghostkeeper, 2020-09-29:
Yeah, I guess it's confusing. I do consider this location to be more logical than what it was before though. Hopefully these are just transitional pains from people who are still used to the old location of the setting. And the improved search function helps.

DreadKnight, 2020-09-29:
Indeed. I've bumped into lots of old video tutorials on YouTube, which didn't help with the issue and nothing while googling :D