Ultimaker/Cura #8388: 3D viewport selection broken on macOS

PhilippMolitor, 2020-09-15:
**Application version**
4.7.1

**Platform**
macOS Big Sur Beta

**Printer**
Ender 3 V2

**Reproduction steps**
1. Open an STL file
2. Trying to select it by clicking the viewport

**Actual results**
Nothing happens. Objects, as well as their rotation/translation gizmos are not selectable.
This issue also seems to affect the preview viewport, which does not show anything at all.

nallath, 2020-09-16:
Please provide the log files and fill in the entire template (we also need things like your GPU, drivers, etc).

PhilippMolitor, 2020-09-17:
Apparently, it only happened to me when I loaded a model from ThingiBrowser, not from a file:
```
2020-09-17 11:45:59,068 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2020-09-17 11:46:02,497 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: RotateOp.(node=<CuraSceneNode object: 'BLTouch_Wing_Narrow_MSDD.stl'>,rot.=Q<0.13052619222005157,0.0,0.0,w=0.9914448613738104>), took 2ms
2020-09-17 11:46:02,603 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: RotateOp.(node=<CuraSceneNode object: 'BLTouch_Wing_Narrow_MSDD.stl'>,rot.=Q<0.13052619222005157,0.0,0.0,w=0.9914448613738104>), took 2ms
2020-09-17 11:46:02,711 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: RotateOp.(node=<CuraSceneNode object: 'BLTouch_Wing_Narrow_MSDD.stl'>,rot.=Q<0.13052619222005157,0.0,0.0,w=0.9914448613738104>), took 3ms
2020-09-17 11:46:02,834 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: RotateOp.(node=<CuraSceneNode object: 'BLTouch_Wing_Narrow_MSDD.stl'>,rot.=Q<0.13052619222005157,0.0,0.0,w=0.9914448613738104>), took 3ms
2020-09-17 11:46:02,983 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: RotateOp.(node=<CuraSceneNode object: 'BLTouch_Wing_Narrow_MSDD.stl'>,rot.=Q<0.13052619222005157,0.0,0.0,w=0.9914448613738104>), took 4ms
2020-09-17 11:46:03,104 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: RotateOp.(node=<CuraSceneNode object: 'BLTouch_Wing_Narrow_MSDD.stl'>,rot.=Q<0.13052619222005157,0.0,0.0,w=0.9914448613738104>), took 3ms
2020-09-17 11:46:04,266 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: RotateOp.(node=<CuraSceneNode object: 'BLTouch_Wing_Narrow_MSDD.stl'>,rot.=Q<-0.13052619222005157,-0.0,-0.0,w=0.9914448613738104>), took 3ms
2020-09-17 11:46:04,511 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: RotateOp.(node=<CuraSceneNode object: 'BLTouch_Wing_Narrow_MSDD.stl'>,rot.=Q<-0.13052619222005157,-0.0,-0.0,w=0.9914448613738104>), took 3ms
2020-09-17 11:46:04,664 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: RotateOp.(node=<CuraSceneNode object: 'BLTouch_Wing_Narrow_MSDD.stl'>,rot.=Q<-0.13052619222005157,-0.0,-0.0,w=0.9914448613738104>), took 3ms
2020-09-17 11:46:04,903 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: RotateOp.(node=<CuraSceneNode object: 'BLTouch_Wing_Narrow_MSDD.stl'>,rot.=Q<-0.13052619222005157,-0.0,-0.0,w=0.9914448613738104>), took 2ms
2020-09-17 11:46:05,264 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: RotateOp.(node=<CuraSceneNode object: 'BLTouch_Wing_Narrow_MSDD.stl'>,rot.=Q<-0.13052619222005157,-0.0,-0.0,w=0.9914448613738104>), took 3ms
2020-09-17 11:46:05,495 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: RotateOp.(node=<CuraSceneNode object: 'BLTouch_Wing_Narrow_MSDD.stl'>,rot.=Q<-0.13052619222005157,-0.0,-0.0,w=0.9914448613738104>), took 3ms
2020-09-17 11:46:05,790 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: RotateOp.(node=<CuraSceneNode object: 'BLTouch_Wing_Narrow_MSDD.stl'>,rot.=Q<-0.13052619222005157,-0.0,-0.0,w=0.9914448613738104>), took 3ms
2020-09-17 11:46:05,929 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: RotateOp.(node=<CuraSceneNode object: 'BLTouch_Wing_Narrow_MSDD.stl'>,rot.=Q<-0.13052619222005157,-0.0,-0.0,w=0.9914448613738104>), took 3ms
2020-09-17 11:46:06,131 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: RotateOp.(node=<CuraSceneNode object: 'BLTouch_Wing_Narrow_MSDD.stl'>,rot.=Q<-0.13052619222005157,-0.0,-0.0,w=0.9914448613738104>), took 3ms
2020-09-17 11:46:07,203 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: RotateOp.(node=<CuraSceneNode object: 'BLTouch_Wing_Narrow_MSDD.stl'>,rot.=Q<-0.13052619222005157,-0.0,-0.0,w=0.9914448613738104>), took 2ms
2020-09-17 11:46:07,925 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: RotateOp.(node=<CuraSceneNode object: 'BLTouch_Wing_Narrow_MSDD.stl'>,rot.=Q<-0.13052619222005157,-0.0,-0.0,w=0.9914448613738104>), took 2ms
2020-09-17 11:46:08,647 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: RotateOp.(node=<CuraSceneNode object: 'BLTouch_Wing_Narrow_MSDD.stl'>,rot.=Q<-0.13052619222005157,-0.0,-0.0,w=0.9914448613738104>), took 3ms
2020-09-17 11:46:08,853 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: RotateOp.(node=<CuraSceneNode object: 'BLTouch_Wing_Narrow_MSDD.stl'>,rot.=Q<-0.13052619222005157,-0.0,-0.0,w=0.9914448613738104>), took 3ms
2020-09-17 11:46:09,858 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: RotateOp.(node=<CuraSceneNode object: 'BLTouch_Wing_Narrow_MSDD.stl'>,rot.=Q<0.13052619222005157,0.0,0.0,w=0.9914448613738104>), took 4ms
2020-09-17 11:46:12,039 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: RotateOp.(node=<CuraSceneNode object: 'BLTouch_Wing_Narrow_MSDD.stl'>,rot.=Q<0.13052619222005157,0.0,0.0,w=0.9914448613738104>), took 3ms
2020-09-17 11:46:13,631 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: RotateOp.(node=<CuraSceneNode object: 'BLTouch_Wing_Narrow_MSDD.stl'>,rot.=Q<-0.13052619222005157,-0.0,-0.0,w=0.9914448613738104>), took 3ms
2020-09-17 11:46:14,045 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: PlatformPhysicsOp.(trans.=<0.0,-6.2,0.0>), took 1ms
[... cut ...]
2020-09-17 11:46:45,754 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: <UM.Operations.AddSceneNodeOperation.AddSceneNodeOperation object at 0x600009f49e58>, took 1ms
2020-09-17 11:46:46,325 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///Applications/Ultimaker Cura.app/Contents/MacOS/resources/qml/ObjectItemButton.qml:69:9: QML Button: Binding loop detected for property "width"
2020-09-17 11:46:56,071 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
```

My GPU is a Radeon Pro 5500M in a MacBook 2019 16".

Ghostkeeper, 2020-09-17:
Hmm, the Thingiverse plug-in fails to load completely for me. Or rather, it loads the plug-in but not the dialogue window:

```
QQmlComponent: Component is not ready
2020-09-17 13:54:08,405 - ERROR - [MainThread] UM.Qt.QtApplication.createQmlComponent [578]: file:///home/trin/.local/share/cura/4.8/plugins/ThingiBrowser/ThingiBrowser/views/Thingiverse.qml:33:5: Type ThingiMain unavailable
2020-09-17 13:54:08,405 - ERROR - [MainThread] UM.Qt.QtApplication.createQmlComponent [578]: file:///home/trin/.local/share/cura/4.8/plugins/ThingiBrowser/ThingiBrowser/views/ThingiMain.qml:15:5: Type ThingiHeader unavailable
2020-09-17 13:54:08,406 - ERROR - [MainThread] UM.Qt.QtApplication.createQmlComponent [578]: file:///home/trin/.local/share/cura/4.8/plugins/ThingiBrowser/ThingiBrowser/views/ThingiHeader.qml:12:5: Type ServiceSelector unavailable
2020-09-17 13:54:08,406 - ERROR - [MainThread] UM.Qt.QtApplication.createQmlComponent [578]: file:///home/trin/.local/share/cura/4.8/plugins/ThingiBrowser/ThingiBrowser/views/ServiceSelector.qml:4:1: Type EnhancedComboBox unavailable
2020-09-17 13:54:08,406 - ERROR - [MainThread] UM.Qt.QtApplication.createQmlComponent [578]: file:///home/trin/.local/share/cura/4.8/plugins/ThingiBrowser/ThingiBrowser/views/EnhancedComboBox.qml:11:5: Cannot override FINAL property
2020-09-17 13:54:08,413 - WARNING - [MainThread] UM.Qt.Bindings.ExtensionModel.subMenuTriggered [58]: Exception: Failed to activate the menu item
2020-09-17 13:54:08,425 - WARNING - [MainThread] UM.Qt.Bindings.ExtensionModel.subMenuTriggered [58]: Traceback (most recent call last):
2020-09-17 13:54:08,425 - WARNING - [MainThread] UM.Qt.Bindings.ExtensionModel.subMenuTriggered [58]:   File "/home/trin/Gedeeld/Projects/Uranium/UM/Qt/Bindings/ExtensionModel.py", line 54, in subMenuTriggered
2020-09-17 13:54:08,425 - WARNING - [MainThread] UM.Qt.Bindings.ExtensionModel.subMenuTriggered [58]:     item["extension"].activateMenuItem(option_name)
2020-09-17 13:54:08,434 - WARNING - [MainThread] UM.Qt.Bindings.ExtensionModel.subMenuTriggered [58]:   File "/home/trin/Gedeeld/Projects/Uranium/UM/Extension.py", line 50, in activateMenuItem
2020-09-17 13:54:08,435 - WARNING - [MainThread] UM.Qt.Bindings.ExtensionModel.subMenuTriggered [58]:     self._menu_function_dict[name]()
2020-09-17 13:54:08,435 - WARNING - [MainThread] UM.Qt.Bindings.ExtensionModel.subMenuTriggered [58]:   File "/home/trin/.local/share/cura/4.8/plugins/ThingiBrowser/ThingiBrowser/ThingiBrowser/ThingiBrowserExtension.py", line 45, in showMainWindow
2020-09-17 13:54:08,435 - WARNING - [MainThread] UM.Qt.Bindings.ExtensionModel.subMenuTriggered [58]:     self._main_dialog = self._createComponent("Thingiverse.qml")
2020-09-17 13:54:08,435 - WARNING - [MainThread] UM.Qt.Bindings.ExtensionModel.subMenuTriggered [58]:   File "/home/trin/.local/share/cura/4.8/plugins/ThingiBrowser/ThingiBrowser/ThingiBrowser/ThingiBrowserExtension.py", line 85, in _createComponent
2020-09-17 13:54:08,436 - WARNING - [MainThread] UM.Qt.Bindings.ExtensionModel.subMenuTriggered [58]:     raise Exception("Failed to create Thingiverse dialog")
2020-09-17 13:54:08,436 - WARNING - [MainThread] UM.Qt.Bindings.ExtensionModel.subMenuTriggered [58]: Exception: Failed to create Thingiverse dialog
```

Maybe has something to do with using a different Qt version than what's used in the build.

I suggest you take this up with the maintainer of that plug-in here, if it only happens with the Thingiverse plug-in: https://github.com/ChrisTerBeke/ThingiBrowser/issues
Perhaps it's got something to do with the QML error of that plug-in's Google Analytics, and this causes the loading of the model to not run to completion.

closed 2020-09-17 by Ghostkeeper