Ultimaker/Cura #8390: [4.7.1] Unsupported material on second layer of tree supports

jtatum, 2020-09-15:
**Application version**
4.7.1

**Platform**
Mac OS X 10.15.6

**Printer**
Creality Ender-3 Pro

**Reproduction steps**
1. Enable tree supports touching buildplate
2. Slice file

**Screenshot(s)**
Layer 1:
[image]

Layer 1 and 2:
[image]

**Actual results**
Note that layer 2 has a new oval shape, with no raft or other supporting material underneath it on layer 1. There are actually two similar areas.

**Expected results**
All supports would extend down to layer 1 (and include rafts where needed).

**Project file**
[IWPLA_triceratops-skull-repair.3mf.zip](https://github.com/Ultimaker/Cura/files/5228767/IWPLA_triceratops-skull-repair.3mf.zip)

**Log file**
N/A

**Additional information**
None.

konskarm, 2020-09-17:
I can reproduce this. I'll put it up for discussion with the team.

Ghostkeeper, 2020-09-18:
Hmm, I couldn't find this myself.
[image]
Where is it going wrong?

konskarm, 2020-09-18:
That's interesting. For me it occurs in the front of the model, under the tip of the jaw. Nevertheless, I couldn't find the exact same spot as @jtatum.

[image]

In your picture, it should be here 
[image]

closed 2024-12-08 by github-actions[bot]