Ultimaker/Cura #8391: Cannot print with Renkforce RF2000 V2 via USB because of missing area

pionsys-mhs, 2020-09-16:
**Application version**
* I am using Ultimaker Cura V 4.7.1

**Platform**
Windows 10 Pro 64 bit

**Printer**
* I selected "Custom FFF printer" but I want to connect to a "Renkforce RF2000 V2"

**Reproduction steps**
1. Open Ultimaker Cura
2. Add a non-networked "Custom FFF printer" and name it "Renkforce RF2000 V2"
3. Connect the printer via USB
4. Switch to "Monitor" view within Ultimaker Cura

**Actual results**
[image]

**Expected results**
The red marked area does not show up in my setup and therefore I cannot print anything via USB.
[image]

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/5230830/cura.log)

Ghostkeeper, 2020-09-17:
It looks like Cura tried to connect to a device on the COM3 port at various baud rates but didn't get a proper response from the printer. As a result it doesn't recognise that your printer is connected via USB cable, and doesn't show any connection in the monitor.

What baud rate would your printer require? Could you try connecting with a different application such as Prontrface to find that out?

pionsys-mhs, 2020-09-17:
I usually use Repetier Host for 3D printing with the Renkforce RF2000 V2. The printer connects with a baud rate of 115200.
My first mistake was t

pionsys-mhs, 2020-09-17:
My first mistake was that i left the Repetier Host connected to the printer while I evaluated Cura and therefore the Cura did nothing.
When I realized i should terminate the active connection before I can connect with another application Cura tried different baud rates (I already saw that within other logs then the attached one) and the light of the printer toggled on/off. The log showed "Access denied" when Cura tried to do some serial writes. Google said I should try to use a different USB port on my machine (Laptops). The "Access denied" error is gone and the light of the printer only flashes one time now but still no connection within the monitor.

Ghostkeeper, 2020-09-21:
Hmm, in the log file it says that it tried a baud rate of 115200 but that didn't make a connection.

pionsys-mhs, 2020-10-02:
Any suggestions on how to resolve this issue?

Ghostkeeper, 2020-10-05:
It could be that another application is holding up the COM port to talk to the printer.

No, I don't really have a workaround for you (or I'd given it already). I don't have a way to debug your problem either without having access to a Renkforce RF2000 V2. And even then I doubt Ultimaker would give me the time to tinker with it, seeing as how they don't sell any printers any more with USB cable printing support.

The best I can do is theorise what happened and hope that someone with access to your printer is able to fix it. And fix it without breaking USB cable printing for other printers.

So you claim that Repetier Host is able to connect to the printer on a baud rate of 115200. But when Cura tries it, it didn't work. Cura uses the [pyserial](https://pypi.org/project/pyserial/) library to connect to the printer. I don't know what Repetier Host uses.
Since it tried 115200 that must mean that either pyserial returned an error that it can't connect at all (e.g. due to the operating system blocking it) or that it connected but got a nonsense response from the printer (e.g. due to bad printer firmware or due to speaking at different baud rates).
Which of those possibilities is happening for you, and how you could fix that, is unknown.

ottle, 2024-11-06:
Hello pionsys-mhs

Ich bin aus Deutschland und versuche gerade ebenfalls meinen RF2000 über USB anzusprechen. 

Bist du auch aus DTL? 
hast Du da noch irgendwas herausgefunden? 

vielen Dank, der Ron