Ultimaker/Cura #8393: [4.7] Inconsistent number of perimeters on bottom layers in vase mode

Liger0, 2020-09-16:
**Application version**
4.6.2, 4.7

**Platform**
Windows 10 x64

**Printer**
Cr10S

**Reproduction steps**
1. Insert a cube
2. Vase mode with bottom layers 

**Screenshot(s)**
[image]
[image]

**Actual results**
The number of perimeter is alternated on each layer

**Expected results**
Same number of perimeters on every layer

**Project file**
[CCR10S_perimeters cube.zip](https://github.com/Ultimaker/Cura/files/5232024/CCR10S_perimeters.cube.zip)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

nallath, 2020-09-16:
I can reproduce the issue. I don't know what exactly is causing this, so I'll discuss it with the team.

Ghostkeeper, 2020-09-17:
Looks like it's on purpose, to make the vase more water tight. It adds a hard-coded extra 5 perimeters.

https://github.com/Ultimaker/CuraEngine/blob/90222b222fc899230efe192111be801f7922da0a/src/WallsComputation.cpp#L29

This code was added in 2013 and never really tracked: https://github.com/Ultimaker/CuraEngine/commit/24c87e639ec8ce879d37423a6cf6dd61ded30cb9

JohnTasto, 2021-08-13:
I was about to open this issue again when I nearly didn't find this one, so allow me to add the keyword _spiralize_ to make it more searchable.

I've been puzzled by this for a while, but it hadn't caused me problems until today. I have a model with a cavity right under where the skin and the extra five walls meet, and the print results in a drawbridge instead of a regular bridge. It prints fine when the cavity is an even number of layers tall, but in this case three layers would really work best if it were only possible.

It would be really helpful if an option existed to either disable it or choose the parity of the layers with extra walls. The existing option `alternate_extra_perimeter` could maybe be piggybacked to disable it, since as far as I know it currently has no effect when `magic_spiralize` is enabled.

UnfoldAntwerp, 2021-11-08:
@Ghostkeeper I don't follow that logic. Spiralize is used a lot in ceramic 3d printing and I don't see much reason that this would make it more water tight. I always use (in Grasshopper or Slic3r/PrusaSlicer) a concentric fill for first bottom solid layer, followed by two or more with parallel lines (the default 90° from each). This makes a great solid bottom. With the market for clay 3d printing becoming less and less of a niche, it's worth putting some thought in this feature. This 'bug' has come up quite a few times in dedicated ceramic fora preventing some people from switching from Slic3r to Cura. So I would plea for either a removal of this undocumented feature (as it can be solved in other ways) or to have it as a setting that can be turned off.

This would be greatly appreciated by all the clay lovers in the audience. THANKS!

Ghostkeeper, 2021-11-12:
Although I wasn't involved yet with Cura back in 2013, I think the reason is that there are often holes or air pockets at the joint between a linear pattern (skin) and the surrounding walls. By alternating the number of walls, the properly watertight walls grasp tightly onto the skin pattern, moving those holes inside to where no water can penetrate.

If you're using concentric skin, that joint doesn't really exist. There is not really a reason to do it then. But that was only the first layer in your case.

Does it hurt though? I understand that it's inconsistent, but hopefully it leads to a strictly better print quality in the end.

UnfoldAntwerp, 2021-11-19:
@Ghostkeeper I can follow that reasoning and I could imagine that feature being useful if tweak able. For plastic printing the 5 extra walls don't add up to much but in clay printing with lines that are often 1-3 mm wide this adds up quickly to fill half the bottom. Anyway, only one extra wall would do the job of filling the gaps in clay, also the issue is not as relevant when using a concentric bottom which is typically waterproof but maybe to thin in plastic? The worry is that in ceramics this feature might introduce unwanted issues as adjacent lines having the same direction over multiple layers may introduce cracks when drying. If you add a concentric to every even layer combined with line infill, you get an area on two opposing sides where all infill lines across all bottom layers (concentric & straight) are parallel-ish. So thats why most of us print concentric + 2-3 diagonal line infill layers to drastically reduce the risk of cracking.

Solution: May I suggest breaking out the _extra walls on alternating layers_ parameter? This way we can adjust the amount of extra concentric layers from 1 to whatever. But more importantly, this would allow us to turn the feature off by feeding it a 0.

ps. on this topic, would it be feasible to take whatever is set under _Bottom Pattern Initial Layer_ also for the top bottom layer if spiralize is on? Concentric is favoured by most of us but due to cracking issues when going all concentric, the workaround is using concentric only for the bottom and parallel lines for the rest as described above. When this would be possible. A 4 layer bottom would be ideal: concentric first layer, two lines layer at right angles, and one more concentric to create a more aesthetic finish on the inside bottom of a cup / vase. That's how I do it in Grasshopper but obviously many people prefer using a slicer instead of hand crafting Gcode :)

Ghostkeeper, 2021-11-26:
That adds an argument for exposing this as a setting. It's not really something I can argue with that this should get priority over other issues though, since ceramic printers are quite rare. But I agree that it could be improved there. I'll have to defer this for now. It'll be delayed until our priorities shift, or until we find more common reasons why one might want this than the current two reasons of inconsistency and cracking in ceramic prints.

> ps. on this topic, would it be feasible to take whatever is set under Bottom Pattern Initial Layer also for the top bottom layer if spiralize is on? Concentric is favoured by most of us but due to [...]

You can do this with a cutting mesh. Make it overlap the 4th layer and change the top/bottom pattern to concentric. Here is a rudimentary example as a project file: [concentric_fourth_layer.zip](https://github.com/Ultimaker/Cura/files/7608334/concentric_fourth_layer.zip) Normally you'd expect the Top Surface Layers to apply to this I think, but internally Cura still thinks it's the bottom of the print even though the infill and top are forced to be 0.

fhoran, 2022-02-09:
@Ghostkeeper I am also using a ceramic 3D printer. If the hard coded value you linked to in your September 2020 comment was smaller it wouldn't cause so many problems for ceramic printing.
I was wondering if I could change the code you linked to in my personal files so that the hardcoded value was lower, at least for me and the students.
I don't know much about code so my apologies if this is the wrong place to ask this.

UnfoldAntwerp, 2022-02-10:
@fhoran maybe we should fork Cura with the ceramic 3dp community. We might be niche but I’m pretty sure we’re with way more than people with the need for printing water tight plastic cups. :)

BeauSlim, 2022-12-22:
I would like to chime in and say this is a problem for me.  Normal FDM printing.  This pattern often makes a mess in the corners of thin (i.e. not tall) rectangular plate-like bases that stick out from the rest of the vase-mode print.  And, if you are forcing bottom lines to print in a single direction (to improve clarity of transparent/translucent filament), this pattern spoils that, too.    Please allow users to turn it off.

Darkgameside, 2023-01-22:
I can also add that this is a issue in FDM printing if you print with 0.6mm+ layer(smaller its not visible) height.
When there are structures around the 'Vase' like lines: ''||''''||''''||''. The filament gets pushed out of the border this leads to a line around the Print.
I tried the provided solutions and other on the internet but they did't worked out. I spend multiple days to configure Cura to prevent this behavoir but the only solution I found is to slice it with Ideamaker. There is the issue not present.

BogusF, 2023-05-20:
@Ghostkeeper 
> Does it hurt though? I understand that it's inconsistent, but hopefully it leads to a strictly better print quality in the end for some prints.

I don't know for others, but it definitly hurts my prints, and they're regular old plastic and none of that fancy ceramic stuff if that helps. ;-) I need rectilinear alignment, so the bottom will come out transparent. 
What hurt more was that it took a lot of time to figure out why cura is doing this and that there aren't any options available to influence it. 

Now the solution/workaround for me will be what I did, when I wanted to have a hole in the bottom of my "vase" and couldn't, because Cura: Slice it twice, once as a solid part and once in vasemode, then manually copy the bottom from the solid part over to the vasemode gcode file and replace the offending bottom. Make sure you either have relative extrusion enabled or insert a "G92 E [new absolute extruder value to continue printing with]" or you extruder might do funny stuff. Also be sure to make the cut at the correct layer height both times.

These hardcoded "the user is dumb so let's be smart for him/her _features_" (there are more of them) really fumble up my workflow every now and then, when i really want to get something right. Then I can't and it takes hours on end to figure things out or find a workaround. It is a major annoyance and the reason why I have been trying to get superslicer to be my new workhorse. Obviously not quite there yet, for better or worse. ¯\_(ツ)_/¯

UnfoldAntwerp, 2024-11-20:
Let me check tomorrow

BeauSlim, 2024-11-21:
Yes, the latest 5.9 (mac/arm) release is still adding these extra perimeters.   
Cura is very much my go-to for vase mode (spiralized) prints and those are more or less my specialty in terms of original designs.  I also use surface-mode (both spiralized and not) a lot.   Cura does things no other slicer does.  Please don't break current functionality!

GregValiant, 2024-11-22:
It changed a lot between 4.13.1 and 5.0 when the variable line width was introduced.  Most of the comments I've heard have been complaints.
I use the Smart Avionics Cura fork for Spiralize.  It is based on the 4.x Cura and I see that his also alternates with 5 wall layers.  It does not depend on "Remove all holes" and so it's easier to put a hole in the bottom of a vase.

GregValiant, 2026-09-07:
I'm cleaning up these old reports.  I will go ahead and close this.
If the problem persists into Cura 5.13.0 please fill out a new report so it goes to the front of the queue.

closed 2026-09-07 by GregValiant