Ultimaker/Cura #8394: [4.7.1] retract into top/bottom layers

userosos, 2020-09-16:
I have a lot of movements on the lower and upper layer. Each blank movement of the Cura makes a retract. As a result, each new line at the bottom or top is accompanied by a retract. After all the lines (see pic 2) cura make retract. See 3mf file also in the zip.
[image]
[image]
[CFFFP_Neat_Jarv.zip](https://github.com/Ultimaker/Cura/files/5232497/CFFFP_Neat_Jarv.zip)

Asterchades, 2020-09-16:
Nothing looks out of place there. Dark blue is just a travel - light blue shows a travel with retract.

userosos, 2020-09-16:
> 
> 
> Nothing looks out of place there. Dark blue is just a travel - light blue shows a travel with retract.

Ok. Thx! I will check my printer. I think that need add into cura visual points for check retract and re-retract pont please!

Asterchades, 2020-09-17:
Can't say that I disagree with you. The distinction between travels is there but it isn't immediately apparent which is which unless you start playing around with settings or someone tells you.

More to the subject at hand, does your printer actually retract at those points or was it just the preview that had you concerned?

userosos, 2020-09-17:
> Can't say that I disagree with you. The distinction between travels is there but it isn't immediately apparent which is which unless you start playing around with settings or someone tells you. [...]

It will bug the printer. I reboot it and ok now. But in cura i want see point retract too.

Ghostkeeper, 2020-09-18:
We've considered adjusting the legend in layer view to show this but never came around to a new design. It's now more likely that we'd implement some new way to see retractions because there is a desire to see Z seams too.

Maybe your printer is moving the feeder for those travels because of Linear Advance?

closed 2024-12-08 by github-actions[bot]