Ultimaker/Cura #8398: IT DON'T WORK! freeze up and crash wont slice correctly this update turned the program into "S**t"

grondi1, 2020-09-17:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

konskarm, 2020-09-17:
Hi @grondi1. I am sorry you are experiencing issues with Cura. Could you please fill in the given template, along with reproduction steps and cura log files?
That way we may be able to reproduce your problems, find out what is causing them and fix them.

grondi1, 2020-09-17:
3 printer profiles created None work! cant connect to Jgaurora A5S Firm ware gone! CR 10 Does what it wants will not print at proper level!Speed control off flow control isn't right even when settings are changed.  NOT happy with Grub Hub Took way to long to make an account!!   In case you didn't get the message" You screwed up the program that was working great now nothing, How can I say THANKS

grondi1, 2020-09-17:
Going back to see if I can get earlier ver that worked

grondi1, 2020-09-17:
Sorry it CRASHED AGAIN!

konskarm, 2020-09-17:
I understand that you are frustrated. However, I am still unsure about the individual problems. 

About the printer profiles missing, I assume you mean quality profiles. There was actually a bug in previous versions which allowed you to see quality profiles that are not compatible with your configuration. This bug is now fixed, which makes the previous (incompatible) profiles to no longer appear.

This fix was made because the profiles that you were seeing were actually not fitting to your printer/material/nozzle combination, which could potentially lead to failed prints due to wrong settings.

Regarding the Jgaurora A5S firmware and CR10 not printing at the proper level, I am not really sure how this is related with Cura.  I will need reproduction steps and project files to investigate the issue, so again, please fill in the given template.

grondi1, 2020-09-17:
OK not going good. how do I get the files if it crashes every time I open?

konskarm, 2020-09-17:
Check this link https://github.com/Ultimaker/Cura#logging-issues. It mentions where to find the logs for each OS.

closed 2020-10-02 by no-response[bot]