Ultimaker/Cura #8399: Horizontal hole expansion every other layer

dekutree64, 2020-09-17:
**Is your feature request related to a problem? Please describe.**
Especially with large (1mm) nozzles, cooling shrinkage can cause the extrusion lines of a hole to split apart from eachother, leading to undersized/out-of-round holes.

**Describe the solution you'd like**
A check box to apply an additional expansion of half the nozzle diameter on every other layer. This makes it so the extrusion paths are laid down half way between the paths on the previous layer, binding them together. This results in the inner wall of the hole having a ridged surface, but that's typically not a problem.

Please ensure that the expanded layers are opposite from the ones that receive an extra wall when "alternate extra wall" is enabled. Otherwise the layers don't stack nicely when that is enabled.

**Describe alternatives you've considered**
The same result can be achieved by building ridges into the model, but it would be much more convenient to do it at slicing time. It adds a great number of polygons, and the ridges must be spaced apart by the exact layer height you plan to print with, and if you need to resize a hole you have to resize the ridges by a different factor to keep the offset equal to half the nozzle diameter.

**Affected users and/or printers**
Mostly users of large nozzles, but it improves hole size accuracy with small nozzles as well.

Ghostkeeper, 2020-09-18:
There are some ongoing experiments with staggering lines at Ultimaker, although not specific to holes. I think making that specific to holes is a bad idea, since holes are not the only place where layer splits occur.

dekutree64, 2020-09-18:
True. Do you have some theory on how it can be done without the stagger pattern showing up on the outer surface? Like maybe extrude the outer wall normally, then extrude the next wall but leave a half-nozzle-width gap between it and the first wall, and then trace along the gap extruding at half flow rate to fill it in, effectively creating a half-width wall. I think that would do the trick.

I'll close out this issue and continue building ridged holes into the model for now, but please do continue working on that :)

dekutree64, 2020-09-18:
On second thought, that half-wall trick probably wouldn't solve the trouble with holes, since the surface wall on each layer will still be stacked directly on top of eachother and would likely split off from the strong interlocked walls.

I really can't think of any way it could be done that won't result in a ridged surface. But since that would be ugly on the outer walls, I do still think a hole-specific solution is the way to go.

Ghostkeeper, 2020-09-23:
You could stagger just the skin and infill lines. There is currently no guarantee that the skin lines fill the shape 100% anyway.
[image]
The walls would still be printed on top of each other then though, so it wouldn't have any effect on thin parts. And with sparse infill it would just make the part weaker.

Another solution could be to print the first inner wall with a different line width (150%?) on alternating layers, and shift the rest of the walls, but I don't think that's a good solution since:
* It requires a big change in flow rate, which will cause underextrusion when increasing the flow rate and overextrusion when reducing the flow rate.
* It will probably reduce the horizontal adhesion between walls.

closed 2020-11-09 by dekutree64