Ultimaker/Cura #8400: OBJ files not importing in Cura 4.7.1 on Windows 10

spartanpace, 2020-09-17:
I'm getting an error every time I try to open or import any OBJ files into Cura 4.7.1. I've never experienced this issue before with any files. I know the files open perfectly in Cura 4.6.2. I use Windows 10. Is there a fix for this bug? Thanks!

[image]

ksihota, 2020-09-17:
OBJ files open fine in my 4.71 on Win10
Have you tried opening the files in 3D Builder or other programs to see if there are any file issues?

spartanpace, 2020-09-17:
Yes. This happens for every single OBJ file I have, which is hundreds of files. I created this file directly from Maya 2020 myself. Again, as mentioned in my original post, I have never had this issue with Cura before until updating to 4.7

spartanpace, 2020-09-17:
Is there something I need to delete in order to create a fresh install of Cura? I've tried uninstalling and re-installing several times

ksihota, 2020-09-17:
To completely uninstall I usually delete the Cura configuration folder under users\AppData\Roaming\cura\version#
This will remove all profiles and changes you have made. Ever major ver (#.#) install, creates a new folder and will attempt to move old data forward. If you delete everything under the cura folder you will be starting fresh. I sometimes just rename my Cura folder to something else so if I need the data I haven't lost it, or just the version folder if I want to bring data forward from the previous version. By doing this I usually don't need to reinstall the application unless there were issues with the original install.
I am having an obj file issue with 4.7 as well, but it is not with loading files for slicing. It is with my obj mesh for my printer machine settings only displaying as white instead of shaded.

nallath, 2020-09-18:
Please fill in the entire bug template. In this specific case, we need a log file (and preferably also one of the files that fails).

Ghostkeeper, 2020-09-21:
Maybe it helps to copy the file to a different location? Like, maybe you don't have the internet connection you require to read from the Google Drive? That's really what the error message indicates. But yes, we'd like to see a log file and a file that fails to load.

fieldOfView, 2020-09-21:
> To completely uninstall I usually delete the Cura configuration folder under users\AppData\Roaming\cura\version#

Pro-tip: if you leave the _empty_ version folder in place (eg 4.7), Cura will not copy configuration files from older versions into that folder but start fresh. If you remove the entire folder and still want to start fresh, you need to remove the folders of older versions too.

ksihota, 2020-09-21:
Thanks fieldOfView. 
That clarification is helpful. I find a lot of users who run into problems when trying to solve issues simply try to reinstall the program files but don't deal with the users configuration folders where all the custom stuff is stored and in many cased is where the problems really exist. I don't think the uninstall previous versions/settings warning/info dialog boxes are clear enough in what they do (when installing and on errors), but most people don't read them anyways,
Now if I can just get my head around the OBJ and image file used to create the machine settings mesh in 4.7 compared to the older versions.

closed 2020-10-02 by no-response[bot]