Ultimaker/Cura #8404: Math error positioning first layer lines

gitcnd, 2020-09-18:
**Application version**
4.7.1

**Platform**
Mac OSX

**Printer**
Anycubic Chiron

**Reproduction steps**
1. Create STL with a line that is exactly 4x the nozzle width, and one layer high.
2. Slice and observe where it places the extrusions.

**Screenshot(s)**
[Screen Shot 2020-09-18 at 3 48 12 pm](https://user-images.githubusercontent.com/4301826/93561039-cf996980-f9c6-11ea-8132-cea831a1ab73.png)

**Actual results**
The last line it draws is not in the right place (slightly overlaps an earlier line, leaving a gap)

**Expected results**
perfect positioning

**Project file**
(forum doesn't accept STL uploads)

**Log file**

**Additional information**
Also behaves badly when creating a line that is 5x the nozzle width: instead of drawing a single straight line in the middle, it attempts to draw a zig-zag in there, which vibrates the printer like crazy and isn't the correct thing to do anyhow.

Ghostkeeper, 2020-09-21:
> *Project file*
(forum doesn't accept STL uploads)

STLs are not project files. You can create a project file by going to File -> Save... The result is a 3MF file with extra metadata about the profiles you were using. In order for Github to accept it, you'll still need to .zip up the project file then.

The AnyCubic Chiron profile sets the Initial Layer Line Width to 140%, probably to improve bed adhesion. As such your line width is not the same as your nozzle size during the first layer. If your model is 4x as wide as the nozzle size then it will not fit exactly. In this case you have two walls overlapping. Wall Overlap Compensation then reduces the flow of one of the lines to prevent it from overextrusion. In practice it tends to fill up the correct area precisely.

We have a better solution in progress here: https://github.com/Ultimaker/CuraEngine/pull/1210

gitcnd, 2020-09-21:
@Ghostkeeper - thanks, but I'm expert.  all initial parameters were set to match later layers (specifically including the initial layer line width to 100% - since I've spent considerable effort getting my bed level to *precisely* the right height, and checked it's correct meticulously [the screenshot is in fact of my level-checking print, which is why the extrusion screwup is annoying me - it's printing it wrong, which is messing my ability to totally perfect the level].).

This is almost certainly some kind of math bug in the code.

Ghostkeeper, 2020-09-23:
Could you show us a project file then? We may be able to figure out what is going wrong then.

gitcnd, 2020-12-10:
Did you even try to reproduce this?  It would have taken you less time to do that, then write asking me to do that for you...

gitcnd, 2020-12-10:
1. Create STL with a line that is exactly 4x the nozzle width, and one layer high.

2. Slice and observe where it places the extrusions.

Ghostkeeper, 2020-12-18:
You want a screenshot as proof? Indeed it looks like the extrusions are perfectly placed:
[image]

This is using a line width of 0.35mm and a model with a width of 1.4mm: [line_width_fitting.zip](https://github.com/Ultimaker/Cura/files/5715978/line_width_fitting.zip)

Please provide a project file for such issues. I could go through all sorts of settings that influence line placement but I'd rather just get all of them at once including the model you're testing with. As you say:

> It would have taken you less time to do that, then write asking me to do that for you...

It is in the issue template as well, precisely for this reason.

closed 2021-01-01 by no-response[bot]