Ultimaker/Cura #8405: [4.7.1] First-run screen upside down

casiosmu, 2020-09-18:
**Application version**
Cura 4.7.1 (x64)

**Platform**
Win10 1909 x64 as VirtualBox virtual machine
Locale de-DE

**Printer**
none yet

**Reproduction steps**
1. Download and install Cura from official website.
2. Run Cura

**Screenshot(s)**
[image]

**Actual results**
Text, text position and button positions vertical inverted, but not the click-area positions of the buttons!

**Expected results**
correct, readable layout

**Project file**
none

**Log file**
No error or exception

**Additional information**
Is this some sort of easteregg?

fieldOfView, 2020-09-18:
> Is this some sort of easteregg?

No, it is an issue with VirtualBox, and does not just affect the first-run wizard. See https://github.com/Ultimaker/Cura/issues/6822, of which this issue is a duplicate.

casiosmu, 2020-09-18:
ok

closed 2020-09-18 by casiosmu