Ultimaker/Cura #8408: Important enhancements in Cura / Eperimental / Wipe Nozzle)

stonefred, 2020-09-18:
Hello developers,

I use the 3D printer "anycubic-predator":
https://www.anycubic.com/products/anycubic-predator-fdm-printer

The filament here is PETG. That doesn't warping. But after a while the nozzle sticks. 

[image]

Which unfortunately slowly but surely destroys the part!

[image]

A cleaning brush cannot be attached to the X-axis (shown in red in the picture) of the delta printer. 
Further optional input values are required:
1. additionally required: Z absolute (shown in green in the picture)
2. additionally required: Y absolute (shown in blue in the picture)
This gives an absolute starting position X-Y-Z for the wipe move (shown in orange in the picture).

Avoid collisions between the nozzle and the printed part: 
If the part to be printed is higher than the entered Z value, cleaning should no longer be carried out.
However, cleaning should be carried out before it is exceeded.

Description of a cleaning procedure for Delta printers:
1. Movement A (magenta)
2. Movement B (red)
3. Multiple wipe movements (yellow)
4. Movement B (red)
5. Movement A (magenta)

The cleaning brush is mounted on the frame of the delta printer and therefore has an absolute position X Y Z (orange).

I think these enhancements are important for users of delta printers.
Cartesian 3D printers also require an additional axis (X or Y).

[Gehaeuse Temp-Steuerung  EMF.zip](https://github.com/Ultimaker/Cura/files/5245123/Gehaeuse.Temp-Steuerung.EMF.zip)

gr5, 2020-09-19:
Note that in your description you assume the brush is always above the part.  You have to do different moves whether the brush is above or below the part.  The brush will start out above the part but if it's a tall enough part, eventually the part will be at the same height as the brush and eventually higher than the brush.

stonefred, 2020-09-19:
Hello gr5,

above I wrote in my entry:
"Avoid collisions between the nozzle and the printed part:
If the part to be printed is higher than the entered Z value, cleaning should no longer be carried out.
However, cleaning should be carried out before it is exceeded. "

With every cleaning procedure, Cura should therefore check whether the currently printed layer is lower than the height (Z absolute) of the brush. If not, the cleaning procedure should no longer be carried out!

[image]

With this rule, the brush can be attached at any height. And there is no risk of collision.
The brush only needs to be mounted over the part.

gr5, 2020-09-19:
@stonefred - you closed this issue by accident.  Please re-open it!

gr5, 2020-09-19:
If the brush is lower than the print, why can't the head move in only XY to be directly over the brush, then go straight down in Z to the brush and move half way back to the part and then go stright up to the most recent Z print height and then start printing again?

stonefred, 2020-09-19:
Hello GR5,

since the inclined rods have no collision sensors, it is not possible to determine whether and when the rods collide with the part.
Therefore, the brush must be OVER the part for safety.

gr5, 2020-09-19:
Ah!  I understand.  The gantry - like portion of a delta can easily hit the part if the printer goes down to hit the brush.  

So in Cura, there are "machine settings" that are designed for an Ultimaker style printer with a gantry that moves the head in X/Y and a bed that goes up and down.  There is also a "head dimensions" set of numbers (two numbers in X and 2 in Y) that define a box around the head.  These are used so you can print in "one at a time" mode where you print something small then go over and print another thing starting with Z=0.  But if the part is taller than the "gantry height" you can't do "one at a time" mode.

So similarly, "gantry height" should be used with this feature and if the brush is lower than the current layer by more than the gantry height (which should be almost zero for you) then it should not wipe on the brush.

gr5, 2020-09-19:
Someone could pretty easily write a post-processing plugin to do all this.  Do you know python?  Are you a programmer?  The elapsed time is written into comments in each layer so you can even parse the time for each layer and when it exceeds some threshold, add a wipe at the layer change.

There are lots of example plugins that you can look at and rewrite.  "pause at layer" is a good simple one.  Probably only about 10 lines of actual code.

stonefred, 2020-09-19:
The spatial position of the inclined bars can hardly be calculated correctly for various delta printers. Much simpler and therefore safer is my suggestion with a brush over the part and a simple collision check if that is not the case.

OK, I have a question about the current implementation in Cura / Experimental / Wipe Version 4.7.1: 
Who implemented these functions?

gr5, 2020-09-19:
@stonefred - you closed the issue again.

stonefred, 2020-09-21:
Hello gr5, thank you for your hints!

gr5, 2020-09-21:
@stonefred - no one is going to look at this issue if you keep closing it.  It is currently closed.  Plus I should warn you that if there is no activity for a certain amount of time (5 weeks?) it will be closed by a bot anyway but certainly if you close it, no one at Ultimaker will ever look at it.

stonefred, 2020-09-21:
Hello gr5, 
I understand. Since this issue is already pretty confused, 
I have now created a **new issue** on this topic:
https://github.com/Ultimaker/Cura/issues/8418

So this issue #8408 can now be closed.
Thank you for your help.

closed 2020-09-21 by stonefred

Ghostkeeper, 2020-09-23:
> OK, I have a question about the current implementation in Cura / Experimental / Wipe Version 4.7.1:
Who implemented these functions?

It was implemented by DesktopMetal. I'm not really content with how it works either, since it's really geared towards their printers rather than a more general case. Maybe it should be made into a g-code field (but then our interface isn't really appropriate for that, and that would also make it harder to use).

Ghostkeeper, 2020-09-23:
> Plus I should warn you that if there is no activity for a certain amount of time (5 weeks?) it will be closed by a bot anyway

FYI, the bot closes issues that have been marked as "Needs info" for 2 weeks, without response from the author of the issue.