Ultimaker/Cura #8409: 4.7.1 won't finish slice with tree supports enabled

pookymaster, 2020-09-18:
**Application version**
4.7.1

**Platform**
Windows 10, 9th gen i5, GTX 1650

**Printer**
Creality CR 10

**Reproduction steps**
1. Loaded STL file
2. Select base .2mm profile
3. Enabled supports, selecting tree supports. 
4. Enabled Coasting
5. Attempted to slice

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
Slicing stalls around 45ish%, never completes. There are no error messages, and the program continues to run. I can cancel the operation slice again with different settings. 

**Expected results**
Successfully sliced object with tree supports

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
I tried disabling coasting to see if that was the problem. It did not fix it. Besides the overhang angle, I did not change any other settings after selecting tree supports.

Vandrasc, 2020-09-21:
Hi @pookymaster , thank you for your feedback! I am not able to reproduce the issue you encountered, could you share the project file with us? Thanks

Ghostkeeper, 2020-09-23:
Probably the same person as here: https://www.reddit.com/r/Cura/comments/iv9kqh/471_wont_generate_a_sliced_file_if_i_select_tree/ (but there is no project file there either). The print is fairly large and it could just be taking a while to compute.

closed 2020-10-05 by no-response[bot]