Ultimaker/Cura #8410: Add Lulzbot 3d Printers

codemonkeyrawks, 2020-09-18:
**Application version**
4.7.1

**Platform**
Debian 10 (Buster) with Intel 600 Series GPU

**Printer**
None

**Reproduction steps**
I would like to be able to submit the lulzbot printers into Cura Ultimaker. The changes that are needed can be found here:

https://gitlab.com/lulzbot3d/cura-le/cura-lulzbot/-/tree/master/resources/definitions
https://gitlab.com/lulzbot3d/cura-le/cura-lulzbot/-/tree/master/resources/extruders

When I try and push a branch I see "fatal: unable to access 'https://github.com/Ultimaker/Cura.git/': The requested URL returned error: 403".

fieldOfView, 2020-09-18:
Only Ultimaker employees can make branches in the Ultimaker/Cura.git repository. The idea is that you create a fork in your own account, create a branch there, and then do a pull request,

Lulzbot made a lot of changes to their LE version of Cura, and because of that the profiles for these printers will not transfer over 1:1 to Ultimaker Cura. Since I nor any of the other regular contributors of Cura have these printers to test with, we cannot vouch for any changes necessary to update the LE files for inclusion with Ultimaker Cura. So someone with access to Lulzbot printers is going to have to do that.

codemonkeyrawks, 2020-09-18:
@fieldOfView I understand exactly. I have a Mini 2 but also new in some regards to porting code into Cura so the best action I might take is creating a bug on "https://gitlab.com/lulzbot3d/cura-le/cura-lulzbot/" for now then. Perhaps at some point Lulzbot will pull some changes in/out and we can go from there. I get using my own modifications are on my own if I use non-lulzbot.

The Ultimaker 5 does look interesting (https://www.matterhackers.com/store/l/ultimaker-s5/sk/MH6DVDNK) but the cost is a bit high for a hobby shop. I have been looking at "https://shop.prusa3d.com/en/3d-printers/180-original-prusa-i3-mk3s-kit.html#" along side of the Mini 2. I guess I am just a little disappointed that the project  here has gone paid (SAS) vs open source fully.

Thank you for the help.

closed 2020-09-18 by codemonkeyrawks

fieldOfView, 2020-09-19:
> I guess I am just a little disappointed that the project here has gone paid (SAS) vs open source fully.

I don't understand what you mean by that. Cura is fully open source, not "software as a service". Can you elaborate what you mean?

codemonkeyrawks, 2020-09-27:
@fieldOfView 

Quote me if I am wrong here but I thought Cura was originally developed along side of Lulzbot and we had one platform to work with at least I thought two years ago we had one code base. I have never updated until recent Cura so possibly my fault here also in not maintaining updates more close. Launching Ultilmaker I see a sign in up top that does imply that the software might be free but looks to add some complexity around managing accounts with Ultimaker directly making it like a SAS at least to me. 

Maybe you can shed some light on why Cura is being maintained by Utilmaker and Lulzbot (separate versions) and maybe what others gain trying to use Utilmaker over Lulzbot Edition. I personally have never used a Ultilmaker printer (cost is quite high) so I have no idea the cost difference between $1000 vs $4000 printer and just trying to figure out the project changes.

I would like to take the feedback you have here and will try and work next with the Debian team to see if we can't package two versions independent as there is some confusion here:

https://packages.debian.org/buster/cura - Utilmaker
https://packages.debian.org/buster/lulzbot-cura (could see) for Lulzbot

As I look at software I just like to try and understand the differences and do get your point with having Lulzbot now Fame 3D add printer support into your platform if chosen. I don;t have a problem here. It just might make it harder for other if there is some features in Ultimaker that are not in the lulzbot edition (personally I don't see any yet but I only have printed for 6 months).

Thank you.

Ghostkeeper, 2020-09-29:
> Quote me if I am wrong here

Cura was originally developed by Daid, then bought from him by Ultimaker. At some point AlephObjects forked Cura and developed a parallel version with their branding. During that time they also provided a number of features for Cura, which were merged back upstream (like the ability to load a .gcode file as layer view). AlephObjects maintained their Lulzbot-branded version for a while. They got stuck at Cura 3.6 though. So any feature of 4.0 or later is not in their version.

Ultimaker offers some software services which are exposed through Cura, like a configuration back-up storage, and printing via Google Cloud. Arguably the main use case of Cura is still offline, but it's a fuzzy border.

codemonkeyrawks, 2020-11-03:
@Ghostkeeper thanks for the reply. I will keep what you mentioned above in mind while trying to work with Ultimaker and Lulzbot while trying to maintain both (any changes I make) apply to both projects.

codemonkeyrawks, 2021-08-29:
@fieldOfView I just wanted to confirm here   that if I made a printer profile in Cura for the Taz 6 and copyed over the start/end  code appropriately this pull would be rejected out of the gate?

I am definitely not impressed with the Lulzbot LE version especially when they don't test packaging releases for Linux. I am asking above because I want to know future wise where to point people if like me where we want to use this version over LE and understand the chances not working with everything.

nallath, 2021-08-30:
> @fieldOfView I just wanted to confirm here that if I made a printer profile in Cura for the Taz 6 and copyed over the start/end code appropriately this pull would be rejected out of the gate?
>
> I am [...]

We have an open pull request for the printers. You can check it out here: https://github.com/Ultimaker/Cura/pull/10232

codemonkeyrawks, 2021-08-30:
@nallath thank you for the confirm! I am actually amazed that Lulzbot has reached out to you for support which is good. 

Will be closing this issue and moving on!