Ultimaker/Cura #8411: Zoom Performance severely degraded in 4.7.1

gleonard3, 2020-09-18:
**Application version**
4.7.1

**Platform**
Ultimaker_Cura-4.7.1.AppImage
Google Chrome: Version 86.0.4240.42 (Official Build) beta (64-bit)
Platform: 13421.32.0 (Official Build) beta-channel nami
Channel: beta-channel
Firmware Version: Google_Nami.10775.123.0
ARC Enabled: true
ARC: 6825202
Enterprise Enrolled: false
Developer Mode: false

**Printer**
CR-6 SE (Custom Add)

**Reproduction steps**
1. Open an .STL.  Attempt to Zoom In or Out.  Note horrible lag.  Basically unusable performance.
2. Verified Panning and ability to Playback Previews is unaffected.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

konskarm, 2020-09-21:
Hi @gleonard3. I am unable to reproduce this issue. Could you please share the cura log files? Also I cannot find the CR-6 SE printer. Could you share that printer profile too?

gleonard3, 2020-09-21:
I wasn't sure how to give you my printer profile, but I created it based of off the stock _Creality Ender 3 Pro_ profile.  To test, I set my activated printer back to the included _Creality Ender 3 Pro_ profile, and the degraded zoom performance was unchanged.  The attached zip contains a cura.log created while opening a Benchy.stl and attempting to Zoom in and out.  I've also included a stderr.log that I thought could be relevant to the issue.

[Issue_8411.zip](https://github.com/Ultimaker/Cura/files/5256643/Issue_8411.zip)

Thanks for looking into this.

konskarm, 2020-09-22:
Thanks for the log files, I'll try to see if there is anything interesting there. You can create a project file (.3mf) with benchy and upload it here. Your printer will be included in that.

gleonard3, 2020-09-22:
I guess I just copied the Ender 3 Pro Printer and called it CR-6 SE.  Anyway, here's the .3mf file.
[3DBenchy.zip](https://github.com/Ultimaker/Cura/files/5261914/3DBenchy.zip)
Thanks again.

konskarm, 2020-09-22:
The provided file is an exported 3mf file (i.e. it includes only the mesh). What we seek here is a project file, which can be generated by (File->Save Project...). Although, indeed, I do not expect it to make much of a difference.

gleonard3, 2020-09-22:
OK, here's another attempt at this:
[3DBenchy2.zip](https://github.com/Ultimaker/Cura/files/5262483/3DBenchy2.zip)

konskarm, 2020-09-23:
As I (and I am guessing you too) expected, log files don't show something out of the ordinary. Same goes with the project file. I notice that you use ChromeOS. Could that be the reason?
Also, could you preproduce the issue in previous versions? Which version did it start appearing first?
Maybe you could attach logs when following the same steps in previous versions, so we can see if there is a difference there.

gleonard3, 2020-09-23:
As a test, I tried the 4.6.2 and 4.7.0 AppImages as well.  The Zoom performance on those versions is only slightly better, bordering narrowly on usable, where 4.7.1 is not at all usable.  

Yes, this is a ChromeOS system which runs Cura in a linux kernel, but you'll note that it is a _Nami_ meaning it supports OpenGL GPU acceleration.  Without this OpenGL support, ChromeOS for Cura becomes much less useful, as you wouldn't be able to playback the Previews, etc.

I also tried PrusaSlicer-2.2.0+linux-x64-202003211856.AppImage, and there are no graphics performance issues on this system with respect to Zooming at all.  Works as expected.

Ghostkeeper, 2020-09-25:
ChromeOS runs those applications in a virtual machine then.

I think you're running into this upstream bug / feature request in the Qt framework: https://bugreports.qt.io/browse/QTBUG-84245 which would indeed severely limit performance.

We didn't really change anything to zooming in 4.7.1 though. We changed some things with updating collision areas when moving or loading models. But that shouldn't happen when zooming. Did it also change when only rotating the camera?

We don't have a ChromeOS computer at work, since it's not supported by us. But that means that we can't really debug this ourselves. Using a virtual machine is supported inasmuch that we assume that the virtual machine emulates a real machine perfectly.

gleonard3, 2020-09-25:
Yes, that is true.  Rotating, Spinning, Playback, all graphic intensive operations work as normal, except Zooming.  It's really a shame, since other than this one issue, Cura works great on this low cost system.  And as I mentioned, PrusaSlicer has no such Zooming issue.  Would be great if you guys could pick one of these up and see for yourself.

Ghostkeeper, 2020-09-29:
Ah yes, I see that Firefox also has no problems scrolling. That's about as useful as a comparison as PrusaSlicer zooming though. They are completely different code bases so it really doesn't matter.
I'm not going to buy a Chromebook just for this bug report. I don't want to have that stuff in my network and I'm not about to blow that money on this bug report either. Ultimaker is not going to buy a Chromebook since it's not supported. But that means that the only thing we can do to fix this bug is to theorise what could cause this issue specifically with zooming.

Maybe it's the scrolling mouse event itself? Does it still perform bad if you zoom by holding the space bar and moving the mouse then? That is a workaround we built once for users without scroll wheel.

gleonard3, 2020-09-30:
My thought process in mentioning PrusaSlicer's zooming capability was to illustrate that reasonable OpenGL accelerated 3D object zooming performance is at least possible on this apparently unsupported platform.  But hey, I'm not a programmer.  Anyway, your space bar suggestion works!  So thank you!  That is a workaround I can live with.  Glad to be able to continue to use Cura going forward.  If you decide this bug is worth pursuing, then I guess the focus is now on the scroll wheel event.  I should mention that this zoom performance bug is also reproducible using two-fingered gestures on the system's track pad, no mouse.

Ghostkeeper, 2020-10-02:
Maybe some Qt flickable element is behaving strangely on that Chromebook then. I wonder what in 4.7 was changed to trigger the difference.

gleonard3, 2020-12-05:
Issue is no longer present running Linux on Chrome OS Version 88.0.4324.22 using Ultimaker_Cura-4.8.0.AppImage.  Presumably the OS update was the cure here.  Thanks.

closed 2020-12-05 by gleonard3