Ultimaker/Cura #8413: Cura 4.7.1 MacOS - no mountable file system [El Capitan 10.11.6]

truegret, 2020-09-19:
**Application version**
Ultimaker_Cura-4.7.1-Darwin

**Platform**
MacOS 10.11.6 - El Capitan
Security & Privacy | Allow apps downloaded from: Anywhere

**Printer**
Creality CR10S

**Reproduction steps**
First attempt
1. In response to "New version available" upon launching of Cura, downloaded 4.7.1 for MacOS
2. Navigated to file location
3. Double-click on Ultimaker_Cura-4.7.1-Darwin.dmg
4. Window to unpack application started
5. Warning window popped up "The following disk image couldn't be opened"
    5.1 Image: "Ultimaker_Cura-4.7.1-Darwin.dmg" Reason: "no mountable file systems"

Second attempt
0. Read release notes 
1. Download Cura 4.7.1 within Chrome browser (https://ultimaker.com/software/ultimaker-cura)
    1.1 Downloaded from Safari - same error
2. Navigated to file location
3. Within System Preferences set Security & Privacy settings
    3.1 Allow apps downloaded from:
    3.2 FROM: Mac App Store and identified developers
    3.3 TO: Anywhere
4. Double-clicked on Ultimaker_Cura-4.7.1-Darwin.dmg
5. Window to unpack application started
6. Warning window popped up "The following disk image couldn't be opened"
    6.1 Image: "Ultimaker_Cura-4.7.1-Darwin.dmg" Reason: "no mountable file systems"

Third attempt
Same results as Second attempt when downloaded using Safari 

**Screenshot(s)**
See attached
Performed in Terminal:
[dgretlein-imac:Ultimaker Cura dgretlein$ hdiutil attach -verbose Ultimaker_Cura-4.7.1-Darwin.dmg 

**Actual results**
1. Window to unpack application started
2. Warning window popped up "The following disk image couldn't be opened"
    2.1 Image: "Ultimaker_Cura-4.7.1-Darwin.dmg" Reason: "no mountable file systems"

**Expected results**
Dialog window allowing me to select where the application is dropped into, and/or with the Applications folder

**Project file**
NA

**Log file**
See attached screenshot

**Additional information**
I successfully launched Ultimaker_Cura-4.7.0
[image]

nallath, 2020-09-21:
This has also been reported on the forums (https://community.ultimaker.com/topic/34080-cura-version-471-for-mac/)

It seems that this is caused by the notarization process. It is something that we are looking into.

Ghostkeeper, 2020-11-17:
This was caused by the notarization process, indeed. In order to notarize, Xcode unpacks the package and re-packages it. The solution was to supply the file system inside the package as a command line argument to the notarization command.

closed 2020-11-17 by Ghostkeeper