Ultimaker/Cura #8415: [4.7.1] Between a Bug and "Feature"

FlailAway, 2020-09-20:
**Application version**
4.7.1

**Platform**
Mint 20+Cinnamon

**Printer**
Any I am guessing

**Reproduction steps**
Save to File

**Actual results**
Open in Text Editor
`
G92 E0
G1 F1500 E-6    
;LAYER_COUNT:55
;LAYER:0
`
**DESIRED, Expected results**
`G92 E0
;  G1 F1500 E-6   // **<-- Stop this line from ever appearing**
;LAYER_COUNT:55
;LAYER:0
`
**Additional information**
On small jobs, I get 20 Skirts because I have set **"Skirt/Brim minimum length = 250** otherwise filament appears at about Layer-2.

Why must there be a mandatory retraction **before** anything is even printed.

Asterchades, 2020-09-21:
Definitely doesn't apply to "any" printer. The default FFF/FDM and Creality profiles don't include such a command in their start GCode and it isn't generated automatically (like heating).

You've either changed your start GCode manually or one (or more) of the profiles needs fixing. But without specifying which printer you're using or uploading a project for inspection it's impossible to tell which is happening for certain.

FlailAway, 2020-09-21:
> Definitely doesn't apply to "any" printer. The default FFF/FDM and Creality profiles don't include such a command in their start GCode and it isn't generated automatically (like heating).

Never say "never," eh! Not meaning to be difficult, but have you actually looked? I just defined an Ultimaker 3 printer, changed **nothing** and loaded an .stl. Sliced and saved the GCode to the desktop and here it is,

`;START_OF_HEADER
;HEADER_VERSION:0.1
;FLAVOR:Griffin
;GENERATOR.NAME:Cura_SteamEngine
;GENERATOR.VERSION:4.7.1
;GENERATOR.BUILD_DATE:2020-09-02
;TARGET_MACHINE.NAME:Ultimaker 3
;EXTRUDER_TRAIN.0.INITIAL_TEMPERATURE:205
;EXTRUDER_TRAIN.0.MATERIAL.VOLUME_USED:2829
;EXTRUDER_TRAIN.0.MATERIAL.GUID:506c9f0d-e3aa-4bd4-b2d2-23e2425b1aa9
;EXTRUDER_TRAIN.0.NOZZLE.DIAMETER:0.4
;EXTRUDER_TRAIN.0.NOZZLE.NAME:AA 0.4
;BUILD_PLATE.TYPE:glass
;BUILD_PLATE.INITIAL_TEMPERATURE:60
;PRINT.TIME:3432
;PRINT.GROUPS:1
;PRINT.SIZE.MIN.X:9
;PRINT.SIZE.MIN.Y:6
;PRINT.SIZE.MIN.Z:0.27
;PRINT.SIZE.MAX.X:135.43
;PRINT.SIZE.MAX.Y:120.991
;PRINT.SIZE.MAX.Z:20.47
;END_OF_HEADER
;Generated with Cura_SteamEngine 4.7.1
T0
M82 ;absolute extrusion mode

G92 E0
M109 S205
G0 F15000 X9 Y6 Z2
G280
G1 F1500 E-6.5                   ; SURE LOOKS LIKE IT WAS AUTO-INCLUDED TO ME
;LAYER_COUNT:198
;LAYER:0
M107`

skarasov, 2020-09-21:
Almost all type of printer.
Maybe remove this firt retract or add an option?

https://github.com/Ultimaker/CuraEngine/blob/90222b222fc899230efe192111be801f7922da0a/src/FffGcodeWriter.cpp#L614

`// retract before first travel move`
`gcode.writeRetraction(storage.retraction_config_per_extruder[start_extruder_nr]);`

Asterchades, 2020-09-21:
The part I find most curious replicated that? There isn't _any_ start GCode listed when I create a new Ultimaker 3 profile. This is before actually slicing anything - just adding the printer and taking a look. And this, at least according to the config file, is by design.

So for reference I tried a different printer. An Anycubic Predator, should you want to verify it yourself. This does have a Start GCode entry. According to the config and in-program dialogue, the start-up code should look as follows:

> G21 ;metric values 
G90 ;absolute positioning 
M82 ;set extruder to absolute mode 
M107 ;start with the fan off 
G28 ;Home 
G1 Z15.0 F1000;short move 
G92 E0 ;zero the extruded length 
G1 F200 E3 ;extrude 3mm of feed stock adjust so I don't have to remove any filament 
G92 E0 ;zero the extruded length again 
G1 F4000 ;move to bed 
;Put printing message on LCD screen 
M117 Printing... 
;G5; this is in order to resume on Power failure (only works on SD prints)

The actual start sequence in the resulting file?

> M140 S65
M105
M190 S65
M104 S205
M105
M109 S205
M82 ;absolute extrusion mode
G21 ;metric values 
G90 ;absolute positioning 
M82 ;set extruder to absolute mode 
M107 ;start with the fan off 
G28 ;Home 
G1 Z15.0 F1000;short move 
G92 E0 ;zero the extruded length 
G1 F200 E3 ;extrude 3mm of feed stock adjust so I don't have to remove any filament 
G92 E0 ;zero the extruded length again 
G1 F4000 ;move to bed 
;Put printing message on LCD screen 
M117 Printing... 
;G5; this is in order to resume on Power failure (only works on SD prints)
G92 E0
G92 E0
G1 F3000 E-4

A few extra lines at the start for temperature and absolute extrusion, which is expected. But not expected are those three lines (one of which is superfluous) at the end - which includes a 4mm retraction. So more digging to find something that corresponds to a 4 for the Predator and a 6.5 for the Ultimaker 3. And there is something.

From the Predator config:
> "retraction_amount": { "default_value": 4 },

From the Ultimaker config:
> "retraction_amount": { "default_value": "6.5" },

Aside from the superfluous quotation marks in the Ultimaker profile, it would appear that it's doing a standard retraction as specified. And, given what skarasov posted while I was typing, it would appear that this is by design, most probably to negate oozing as it moves from priming into position. This would explain why I never noticed anything untoward since my retractions are only 0.4mm, and I actually have it performing one by design in my start-up script.

skarasov, 2020-09-21:
It's not start script. It's hardcoded in Engine.

Asterchades, 2020-09-21:
Yes, so I alluded to.

I can see this being a potentially significant issue in bowden systems, given there's a retraction performed automatically at the end of a print as well. It won't make a huge difference if you're only getting ~1mm of retraction, but if you're using a long tube with 6mm or more at both the end of one print and the start of the next? Yeah, that's a problem, and may explain why my pre-direct drive testing showed no alternative but to use a purge line.

FlailAway, 2020-09-21:
> Yes, so I alluded to.

Shall I take that to be an apology for the somewhat demeaning and autocratic response to my OP?

Asterchades, 2020-09-22:
No, but you are quite right that you deserve one from me. I still stand by my assertion that it isn't adding a 6mm retract on every printer, but it clearly is automatically adding a retraction based on settings given either by the printer profile or material profile as per the code snippet referenced by skarasov.

An argument could be made that such a retraction would be a reasonable idea (though not essential) if a purge line is used. But as not every printer uses a purge line by default it's going to cause problems, meaning it would be better suited to being added manually to the start GCode following said purge instead of automatically to every output.

Ghostkeeper, 2020-09-23:
> those three lines (one of which is superfluous) at the end

Yeah the superfluous one is a long-standing [bug](https://github.com/Ultimaker/Cura/issues/6957). It's not seen as a big problem but it's indeed unnecessary.

I'll close this as a duplicate of https://github.com/Ultimaker/Cura/issues/5713. This retraction is written since the move to the first position is always a move outside of any objects, so ooze is important. It's expected that the filament is at the nozzle tip at the end of the start g-code, but if you also retracted in the g-code itself it'll stay retracted of course.

closed 2020-09-23 by Ghostkeeper